rusefi/rusefi #4359: ttyACM* gets disturbed by ModemManager on modern Linux distro

dron0gus, 2022-07-20:
CDC ACM is interface mostly used for modems.
ModemManager tries to detect modem on all newly added ttyACM devices.
This breaks TunerStudio communication few seconds after establish when ModemManages sends few AT commands.
https://stackoverflow.com/questions/24696527/modem-manager-and-ttyacm-in-use

Simple solution:
`systemctl stop ModemManager.service`

Advanced solution:
Add VID, PID to udev rules with `ENV{ID_MM_DEVICE_IGNORE}="1"` attribute
https://linux-tips.com/t/prevent-modem-manager-to-capture-usb-serial-devices/284/2

dron0gus, 2022-07-20:
```
$ systemctl status ModemManager.service
● ModemManager.service - Modem Manager
   Loaded: loaded (/lib/systemd/system/ModemManager.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2022-07-20 11:50:04 MSK; 8h ago
 Main PID: 1242 (ModemManager)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/ModemManager.service
           └─1242 /usr/sbin/ModemManager --filter-policy=strict

июл 20 20:02:05 DronG ModemManager[1242]: <warn>  (ttyACM0) could not open serial device (2)
июл 20 20:02:05 DronG ModemManager[1242]: <warn>  [plugin manager] task 93,ttyACM0: error when checking support with plugin 'Cinterion': '(tty/ttyACM0) failed to open port: Could not open serial device ttyACM0: 
июл 20 20:02:05 DronG ModemManager[1242]: <warn>  (ttyACM0) could not open serial device (2)
июл 20 20:02:05 DronG ModemManager[1242]: <warn>  [plugin manager] task 93,ttyACM0: error when checking support with plugin 'u-blox': '(tty/ttyACM0) failed to open port: Could not open serial device ttyACM0: No 
июл 20 20:02:05 DronG ModemManager[1242]: <warn>  (ttyACM0) could not open serial device (2)
июл 20 20:02:05 DronG ModemManager[1242]: <warn>  [plugin manager] task 93,ttyACM0: error when checking support with plugin 'Generic': '(tty/ttyACM0) failed to open port: Could not open serial device ttyACM0: No
июл 20 20:02:05 DronG ModemManager[1242]: <info>  Couldn't check support for device '/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.8/2-1.8.3/2-1.8.3.3': not supported by any plugin
июл 20 20:02:25 DronG ModemManager[1242]: <info>  [device /sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.8/2-1.8.3/2-1.8.3.3] creating modem with plugin 'Generic' and '1' ports
июл 20 20:02:25 DronG ModemManager[1242]: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'
июл 20 20:02:25 DronG ModemManager[1242]: <warn>  Couldn't create modem for device '/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.8/2-1.8.3/2-1.8.3.3': Failed to find primary AT port
```

mck1117, 2022-07-20:
does hardware CI run this service?

let's find out:

https://github.com/rusefi/rusefi/pull/4361

dron0gus, 2022-07-20:
Do we have reconnect issues on CI?

mck1117, 2022-07-20:
nope

chuckwagoncomputing, 2023-01-28:
Can we add a note in the wiki, maybe link it from https://wiki.rusefi.com/HOWTO-quick-start/, and close this? Unless there's some actionable firmware change, but I don't see that there is.

chuckwagoncomputing, 2023-01-29:
Somebody else with this issue
https://github.com/qmk/qmk_firmware/issues/3804

Adafruit tells people to just remove ModemManager
https://learn.adafruit.com/adafruit-arduino-ide-setup/linux-setup#udev-rules

chuckwagoncomputing, 2023-02-23:
This is documented now, I suggest we close this issue unless we plan on doing something within the firmware to prevent this from happening, which I don't think is feasible.

closed 2023-02-23 by rusefillc