rusefi/rusefi #4373: What about "catch all CAN" via Lua

rusefillc, 2022-07-22:
Do we have the API to CAN RX any message not just the speciifc list?

for instance vaguely related to crazy MS 29 bits https://github.com/rusefi/rusefi/blob/master/firmware/controllers/lua/examples/ms29bit.txt or just cases when we need to relay 100s of different messages from bus one to bus two while fildering our very few IDs or just changing baud rate

rusefillc, 2022-07-22:
@NMSTEC FYI

kjngineering, 2022-07-22:
Proposal from https://github.com/rusefi/rusefi/issues/4374

could work like:

canRxAdd(0x20000000,0xFFFFFFFF,*callback)

which would be a capture all for all addressses.

mck1117, 2022-07-26:
> canRxAdd(0x1FFFFFFF,0x1FFFFFFF,*callback)
> 
> which would be a capture all for all addresses in standard + extended.

or canRxAdd(0, 0, callback)

PR #4387 (2022-07-28): Lua rx callback
- Filter by bus in addition to ID (only rx from one bus instead of all)
- Add option to filter with a mask, so you can subscribe multiple messages with one call (or subscribe all messages)
- Allow passing a callback to be called instead of the generic catch-all callback.

fix #4320
fix #4373
fix an alternative to  #4374
closed 2022-07-28 by rusefillc