rusefi/rusefi #4374: canRxAdd to accept array as well as singe ID

rusefillc, 2022-07-22:
More or less API sugar suggestion to replace

```

canRxAdd(0x18B838)
canRxAdd(0x58B838)
canRxAdd(0x60B838)
canRxAdd(0x48B838)
```

with

```
canRxAdd(0x18B838, 0x58B838, 0x60B838,0x48B838)
```

rusefillc, 2022-07-22:
@NMSTEC FYI

kjngineering, 2022-07-22:
as discussed on slack:

instead of a function like:
`canRxAdd(0x18B838, 0x58B838, 0x60B838,0x48B838)`

I propose something more inline with how other can libraries deal with filtering:
`canRxAdd(0x03838,0x87FFF,*callback)`
and/or
`canRxAdd(bus, 0x03838, 0x87FFF, *callback)`

so if the message matches 0x03838 & 0x87FFFF call the function; callback.

@mck1117 "We get a function called by name, then just inspect the stack to handle whatever args you passed"

mck1117, 2022-07-28:
@rusefillc do we think this is closed by #4387 by using a mask instead of a list?

rusefillc, 2022-07-28:
@mck1117  think those are two separate use-cases

for instance for BMW TCU man-in-the-middle two explicit lists would be better than a mask

kjngineering, 2022-07-28:
@rusefillc what are the BMW TCU frames? you could have multiple filters calling the same callback function?

rusefillc, 2022-07-28:
https://github.com/rusefi/rusefi/blob/e459300897a9730a8132b8ec7a8e1ae68a77501e/firmware/config/engines/bmw_n73.cpp#L62

mck1117, 2022-07-28:
Remember that it applies all filters in order, so if you wanted to receive all messages EXCEPT one, you could first filter that one to a no-op callback with no mask, then a catch-all of every message.

If you're doing an MITM, why not just grab every frame?

mck1117, 2022-09-05:
are we happy with the current combination of rx apis available?

rusefillc, 2022-09-06:
I still see
```
canRxAdd(0x18B838, 0x58B838, 0x60B838,0x48B838)
```

as a separate legit use-case