rusefi/rusefi #4382: Martin reports that latest snapshot bricks his MRE

rusefillc, 2022-07-26:
Martin reports that latest snapshot gives him FATAL without USB

rusefillc, 2022-07-26:
https://rusefi.com/online/view.php?msq=1131

rusefillc, 2022-07-26:
0715 works
latest fatal

would be nice to isolate date of issue

See https://github.com/rusefi/rusefi/actions/workflows/build-firmware.yaml?query=branch%3Amaster

each link like https://github.com/rusefi/rusefi/actions/runs/2720877025 has artifacts - use MRE files from artifact

rusefillc, 2022-07-26:
from experience lack of USB means ChibiOS halt like #3186

rusefillc, 2022-07-26:
well we have #2859 and yet it did not help

mck1117, 2022-07-26:
I'll have a look tonight with a debugger attached

rusefillc, 2022-07-26:
so far I am unable to reproduce

```
[4:23 PM, 7/26/2022] ecu martin honda: hmmm... not so happy because i cant see, ohh shit
[4:23 PM, 7/26/2022] ecu martin honda: i unsoldered board with sd card
[4:23 PM, 7/26/2022] ecu martin honda: can this kill usb?
```
it's the unofficial MRE board

[image]

mck1117, 2022-09-05:
no repro, close?

closed 2022-09-05 by rusefillc