rusefi/rusefi #4413: Code Reporting for Voltage during Cranking

BeerMoneyMotorsports, 2022-08-03:
During cranking, we expect X amount of voltage to have a properly functioning ECU.  Below X voltage, we would have inconsistent or no startup.  Reporting "Battery below X or minimum voltage during cranking" as an error would help identify this sooner and assist in overall user experience.  

I have seen this on multiple occasions with Miata.  Old batteries in a project that has sat around a while. My personal car, the lemon, shows this in certain situations after sitting a few weeks.  

Could this be part of codes to be reported also through CEL activation?  For the instances that Tuner Studio is not able to show the Error?

mck1117, 2022-08-03:
what behavior do you see on your car when the battery is low? enough cranking speed that you expect it to start but it doesn't?

mck1117, 2022-08-03:
X is probably lower for rusEFI than the starter will turn the engine over.

BeerMoneyMotorsports, 2022-08-03:
For my car, it will crank but not start. Slower cranking but still cranks. 

For customers, ive seen it drop to 7/8v during cranking and it wont start.  Seeing 200-250rpm while cranking. Toss a jump pack on and it fires right up. I will attempt to log it.

mck1117, 2022-08-03:
We're fully functional down to about 6. I'd expect the ECU to be able to start the car at 6. Grab a log and let's see what's up.

BeerMoneyMotorsports, 2022-08-03:
After matg.

duhafnusa4, 2022-08-04:
I have had this issue on my car as well

mck1117, 2022-08-04:
Logs logs logs!

rusefillc, 2022-08-08:
mre-adapter-for-NB2 canned tune not sure if even functional https://rusefi.com/online/view.php?msq=1149

dead battery with 4 minutes of 50 amps charge as first log https://rusefi.com/online/view.php?msq=1149

rusefillc, 2022-08-19:
https://rusefi.com/online/view.php?log=970

rusefillc, 2022-10-03:
Not done