rusefi/rusefi #4420: BRAND_NEW_UNIT flag and fatal error to facilitate initial setup

rusefillc, 2022-08-06:
problem statement: initial setup
contributing factor: warnings are invisible

solution:  fatal errors have great visibility with clear text
nuace: we do not want to fatal while in the corner at the track

proposed solution:
* new BRAND_NEW_UNIT flag
* special handing of BRAND_NEW_UNIT: fatal error if SD selected AND one second since boot AND map value out of 100kpa range
* special handing of BRAND_NEW_UNIT: fatal error after 120 seconds of running to force user to turn off BRAND_NEW_UNIT flag manually

[image]

rusefillc, 2022-08-06:
@BeerMoneyMotorsports @stefanst opinions?

mck1117, 2022-08-07:
> special handing of BRAND_NEW_UNIT: fatal error if SD selected AND one second since boot AND map value out of 100kpa range

Having seen cars get limped in off-track after hardware failures/crashing/a spin/whatever (and an ignition cycle!), I'm not a fan of this logic at all. I've also rebooted the car while moving, in gear, making vacuum. My cars also start in far less than a second.

mck1117, 2022-08-07:
It seems like this is not worth adding additional new complicated and probably annoying feature instead of writing an accurate and useful "before first start checklist" document.  I am sure several of us would be happy to contribute content to such a document.

rusefillc, 2022-08-07:
@mck1117 I am looking at https://github.com/rusefi/rusefi_documentation/issues/227 and it tells me everything I need to know about managing documentation content in the open source setup

more importantly I believe that average users would not read any documentation, including our existing https://github.com/rusefi/rusefi/wiki/First-Engine-Start which has no content relevant for this issue here

rusefillc, 2022-08-07:
> Having seen cars get limped in off-track after hardware failures/crashing/a spin/whatever (and an ignition cycle!), I'm not a fan of this logic at all. I've also rebooted the car while moving, in gear, making vacuum. My cars also start in far less than a second.

please clarify if I was unclear with "special handing of BRAND_NEW_UNIT: fatal error after 120 seconds of running to force user to turn off BRAND_NEW_UNIT flag manually" or if you did not read that far?

ElDominio, 2022-08-09:
Sadly, I don't think this is fixable. Even the best-of-best ECUs always say in the manual RTFM, the program annoys you with RTFM, and no one does RTFM.

I think the most useful way to do something like this is to put an onboard piezo buzzer that doesn't shut up until manually disabled lol

rusefillc, 2022-08-09:
[image]