rusefi/rusefi #4421: some ignition events skipped

mck1117, 2022-08-06:
Some cylinders seem to not fire even though sync looks OK.

Sequential mode:

Ignition 2/4 missing, but those cylinders get fuel.

[image]

Wasted spark:

<coming soon>

Log: https://rusefi.com/online/view.php?log=946
Tune: https://rusefi.com/online/view.php?msq=1137

mmatousek89, 2022-08-12:
seems like you are locked in 0-360degree of engine cycle and 360-720 is ignored .

rusefillc, 2022-08-31:
@mck1117 how shall we proceed in your opinion? what data do we have or need to reproduce this issue?

mck1117, 2022-08-31:
I can have a look at this to try and reproduce with the provided info, I think it's enough to at least try.

mck1117, 2022-10-11:
I think this is *probably* fixed. We saw a few people with similar issues, and updating to latest seemed to fix the issue.

mck1117, 2022-12-01:
I think it's fixed, but we can reopen if we see it again.

closed 2022-12-01 by mck1117