rusefi/rusefi #4424: 4B11 VVT issues: positive duty makes Exhaust VVT go negative

ElDominio, 2022-08-09:
As title says, that happens, but VVT angle cannot be negative. Need to target negative angle since this is expected operation

rusefillc, 2022-08-09:
duty? angle? I am very confused

tune URL is best

ElDominio, 2022-08-09:
https://rusefi.com/online/index.php?vehicleName=Lancer+NotEvo+fifrst&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

Increasing exhaust VVT duty makes the VVT angle go into negative numbers (when idling at 0 degrees)

rusefillc, 2022-08-09:
@ElDominio what is "actual behaviour" and what is "expected behaviour"

I am starting to wonder if that's about VVT target table not allowing negative values

Are you asking for VVT target to start accepting negative values?

[image]

ElDominio, 2022-08-09:
Correct, that would be best

rusefillc, 2022-08-09:
just a reminer that you would need to lower PID min value to include your negative range

mck1117, 2022-08-09:
there's a other step, the control is reversed for his exhaust cam

rusefillc, 2022-08-09:
For reversed control do we simply use negative P?

mck1117, 2022-08-09:
> For reversed control do we simply use negative P?

It's much more user friendly to not do that.

mck1117, 2022-08-09:
> just a reminer that you would need to lower PID min value to include your negative range

PID min is the min output - not the min target

rusefillc, 2022-08-09:
https://github.com/rusefi/ts-issues/issues/52

ElDominio, 2022-08-14:
https://rusefi.com/online/view.php?log=967

VVT exhaust angle target is negative, PID max and min are 90 and -90 respectively. No change in VVT angle, VVT direction for exhaust is set to "retard"

https://rusefi.com/online/view.php?log=968

This log has VVT direction set to "advance", and once the target is changed to a positive value(at about 30s), actual VVT goes rapidly towards -40 (max VVT angle on exhaust), I assume because the PID is seeing it get farther and farther away from +10 (the target) .

ElDominio, 2022-08-20:
👀

commit 97601f92 (2022-08-09): VVT target to allow negative values fix #4424
PR #4464 (2022-08-21): fix inverted vvt
fix #4424
closed 2022-08-21 by rusefillc