rusefi/rusefi #4427: ISO-TP does not work on real car

rusefillc, 2022-08-09:
ISO-TP works on bench but not on real car

rusefillc, 2022-08-09:
UDS "3Eh Tester Present" could help lower CAN bus traffic on the vehicle

http://mdac.com.au/2019/06/02/diy-ecu-reprogramming-by-can/

Roman says CAN id = 0x7DF

shadowm60, 2022-08-09:
What is the complete use case?

For diag there is physical and functuonal diag ids configured, one is broadcast and com control service 28h would transmit to disable tx, enable tx, this way communication is reduced, then for the target ecu enable tx and enable tx on the same 28h service but now on physical can id so only thqt one would react.

Iso tp for some oem.s do have st.min and other params configured by default or on the fly

If there is a diag req ongoing on an isotp channel, do not send anything else on the same diag channel, not even tester present, that will drop initial request and process tester present.

rusefillc, 2022-08-09:
We have TunerStudio protocol wrapped into ISO-TP so rusEFI ECU and Windows TunerStudio can talk via ISO-TP+java code + PCAN dongle and PCAN drivers https://rusefi.com/forum/viewtopic.php?f=5&t=2233

this works on a bench with only rusEFI present

Same hooked into my 2011 Xterra does not work via OEM vehicle OBD CAN bus. I know that OBD on 2011 xterra reads ECU traffic. Rest of the vehicle seems to be producing about 1100 messages per second, at the moment I am investigating if those 1100 extra messages are killing what laters of rusEFI CAN TunerStudio.

shadowm60, 2022-08-09:
In this case, is the obd port connected to engine via a gateway ecu?

Did you tested obd connected to engine can?

If there is a gateway, usually there is routing, and a protocol to set up car to accept iso tp traffic for reflash, etc. 

I will also check forum, but i would test with engine can directly and also check can load, isotp could work with high bus load as well, so i see no reason why, maybe can trace from engine can and from diag port to compare traffic? And timing?

rusefillc, 2022-08-09:
https://github.com/rusefi/rusefi/commit/7b5eedf6bc215e15843167e07f35dd66728a52f5 is a step toward confirming that packet has transmitted from PC to ECU via vehicle

rusefillc, 2022-08-09:
OBD definitely reads rusEFI messages based on:

ignition OFF
with rusEFI ECU not hooked to USB I have silence on the CAN bus as reported by rusEFI PCAN java client

once I hook rusEFI ECU to USB i see ~450 CAN messages per second, all originating from RE

in the meantime ISO-TP message does not reach RE based on USB isoTpVerbose logging

rusefillc, 2022-08-11:
OMG it was soooo lame : Nissan binary had ISO-TP code disabled on preprocessor level :(

commit 17080df0 (2022-08-11): ISO-TP does not work on 2011 Xterra fix #4427
closed 2022-08-11 by rusefillc