rusefi/rusefi #4428: microRusEFI doesnt connect as mass storage device

mmatousek89, 2022-08-09:
MRE doesnt connect as mass storage on one computer but it does on other computer.

rusefillc, 2022-08-09:
@mmatousek89 just to confirm do you mean
a) the "your ini file and links" totally virtual drive
or
b) the SD card viewing MSD?

mmatousek89, 2022-08-09:
both doesnt show up , only STMicroelectronics Virtual COM Port  (COM26) apear in device manager

rusefillc, 2022-08-09:
this has came up more than once we need better info on this :(

rusEFI presents itself as up to three devices within a composite USB device to Windows

I recall there are steps to view composite device detailed into in Windows but I do not recall those exact steps

mmatousek89, 2022-08-09:
it happen after i brick my other MRE an try to fix it via instaling drivers from rusefi console, sinc then mine MREs doesnt show as mass storage device nor virtual with ini nor pseudovirtual representing SD card

mmatousek89, 2022-08-09:
ok mine fix for this issue is to go to device manager and uninstal device check checkbox to uninstal device related software and than disconect/conect MRE back to computer and all storages show up again, and virtual com port now use diferent driver called "serial device USB (COM9)" now i need to test if tunerstudio still works and allso maybee firmware update can be affected... i dont know yet

mmatousek89, 2022-08-09:
as far as i use allso win7 machine with disabled updates , i tried removing STMicro virtual com too, but now win7 cant find driver for rusefi ecu, it says that its rusefi ecu in device manager and i can see virtual mass storage with ini file and removable device for sd card

on win10 windows asigned some native/universal driver and tunerstudio is working fine allso mass storage is working fine

so "instal drivers" from RusEFI console seems no go for me.... i tried to instal drivers again and stor comes to begining so i can replicate this situation if needed for some debug

mck1117, 2022-08-09:
Windows 7 doesn't support the ini/sd drives at all. Only windows 10 and later are supported, because it has the generic serial port driver you mention.

rusefillc, 2022-08-09:
OMG this ticket is now totally detroyed :(

a) original description does not spell out windows version
b) we've started to mix up win 7 and 10. at this point this ticket is more confusing than I can handle :(

mmatousek89, 2022-08-09:
right now it seems like only think needed to make it work on win7 is to somehow instal usbser.sys to win7 but its probably not worth the hassle and im not clever enough to do it

https://www.microchip.com/forums/m564739.aspx

https://stackoverflow.com/questions/41928144/inf-file-cant-find-usbser-sys-in-windows-7-only

here are probably related links

mmatousek89, 2022-08-09:
behaviour is allmost same on win7 and win10 , driver instal from rusefi console simply kill massstorage functionality and allow to use only virtual com port. 

on win7 driver is necesary to have acess to virtual comport so without any workaround i have to sacrifice mass storages

on win10 it seems that system can instal its native driver for virtual com and mass storages , so instaling driver from rusefi console only hurt things because result is lost of mass storage functionality.

mck1117, 2022-08-09:
> on win10 it seems that system can instal its native driver for virtual com and mass storages , so instaling driver from rusefi console only hurt things because result is lost of mass storage functionality.

I think this is the valuable nugget of information here.

Let's ignore win7 for now.

rusefillc, 2022-08-17:
stale and mixing unsupported win7 with supported win10

proper signal is to document that win7 is totally not supported for MSD but do not see resources for that.

closed 2022-08-17 by rusefillc