rusefi/rusefi #4431: acceleration enrichment default values

rusefillc, 2022-08-10:
@blundar says code is amazing but default values are horrible

We need much more sane defaults

rusefillc, 2022-08-10:
current default probably https://rusefi.com/online/view.php?msq=1142&dialog=AccelEnrich

@blundar says WW is not taking care of 100%
@blundar says that somehow @mck1117's car is the only one with WW doing 100% of the job open question why

rusefillc, 2022-08-10:
@BeerMoneyMotorsports how are things with accel enrichment with miata? please provide a URL of the happiest tune in this department

rusefillc, 2022-08-10:
open question if this somewhat related to #3940

idi-moe, 2022-10-17:
Default values are indeed horrible. A few reasonable defaults I think would be:
Length: 0.10sec
Accel Threshold: 2.0
Decel Threshold: 2.0 (or less perhaps? haven't messed with this that much)

I am unsure if the accelerator pump model should have default values, as they seem to be rather dependent on everything else.

Also, the default TPS/TPS map has axes of 0,10,30,40,60,70,90,100. This spacing is not good. I propose that instead the default axes follow the throttle positions at regular intervals of flow, with perhaps a bias for sampling more at lower areas.

I will be experimenting with this and update in a week or two hopefully.

[http://www.diva-portal.org/smash/get/diva2:23184/FULLTEXT01.pdf](url) <-good paper on throttle valve modelling

BeerMoneyMotorsports, 2022-10-17:
The lemon seems OK but not amazing.  Using the advanced Wall wetting tables has made an improvement.

https://rusefi.com/online/view.php?msq=1267

idi-moe, 2022-10-17:
> The lemon seems OK but not amazing. Using the advanced Wall wetting tables has made an improvement.
> 
> https://rusefi.com/online/view.php?msq=1267

Are said tables viewable in rusefi online? Can't seem to see them.

rusefillc, 2022-10-17:
@idi-moe please paste a URL of a tune with better proposet axis values, or just give me 8 values instead of 0,10,30,40,60,70,90,100

idi-moe, 2022-10-17:
> @idi-moe please paste a URL of a tune with better proposet axis values, or just give me 8 values instead of 0,10,30,40,60,70,90,100

I'm going to be coming up with some in the next two weeks after testing and modelling, will certainly post link then.