rusefi/rusefi #4433: VVT position is too jumpy

rusefillc, 2022-08-11:
@ElDominio has surviving at 750ish idle, at 900 idle we expect it to be even happier

yet at https://rusefi.com/online/view.php?log=958 we see vvt position just from 3 to -8 in one cycle with VVT solenoid unhooked

why? looks like VVT position should be way smoother

rusefillc, 2022-08-11:
@ElDominio yep, it's based on time since sync and on different vehicles angular distance between crank sync point and VVT sync point could be different causing different precision

```
/**
 * @return angle since trigger synchronization point, NOT angle since TDC.
 */
expected<float> TriggerCentral::getCurrentEnginePhase(efitick_t nowNt) const {
	floatus_t oneDegreeUs = engine->rpmCalculator.oneDegreeUs;

	if (cisnan(oneDegreeUs)) {
		return unexpected;
	}

	return m_syncPointTimer.getElapsedUs(nowNt) / oneDegreeUs;
}

```

mck1117, 2022-08-12:
@rusefillc - he mentioned this engine already has a lumpy cam that causes some significant speed variation at idle.

[image]

It may be worth updating engine phase at every primary trigger tooth, instead of just using the sync point alone.

mck1117, 2022-08-12:
So any variation in engine speed between the sync point and VVT tooth show up as jitter in the resolved VVT position. With a lumpy cam, you get a lot of that speed variation.

rusefillc, 2022-08-12:
@ElDominio reports amazing progress https://rusefi.com/online/index.php?vehicleName=raymond&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

thank you @mck1117

ElDominio, 2022-08-12:
The test car will soon be driven around the block, and the test car in Canada will also be updated today. If all is well boost will start being tuned too

ElDominio, 2022-08-12:
Test car 1 (Forte) driven around the block, owner reports no random shutdowns, car turns on perfectly

Test car 2 (Genesis) driven up to gas station, owner reports YES random shutdowns, car turns on most of the time. I see the exhaust cam doing that weird reading correctly-then jumping between both ends of positive and negative. I will try to have logs somewhat soon.

ElDominio, 2022-08-12:
https://rusefi.com/online/index.php?vehicleName=Guelph+Genesis&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

Shows cranking VVT angle for exhaust jumping between positive and negative. Once it stabilizes at 0, it runs perfectly and doesnt fuck up again

rusefillc, 2022-08-12:
I am concurned that we might we mixing different issues here. We need to establish causes and cosequences

Root cause could be closer to #4437

ElDominio, 2022-08-12:
But this is while cranking, it doesn't start until angle shows 0. If it stays at 0 on the first spin, it starts right up.

rusefillc, 2022-08-12:
This ticket is about VVT position being jumpy while engine is running. Have we made progress? Have we resolved "vvt jumpy while idling"? Can we close this issue?

ElDominio, 2022-08-12:
Yes

closed 2022-08-13 by rusefillc