rusefi/rusefi #4436: 1995-2003 VWgroup idle throttle jack controll

mmatousek89, 2022-08-12:
vw/audi/seat/skoda  used od 1.8T 20V and VR6 engines older style of throttlebody with "idle jack" which is basicaly DBW mechanism which can controll 0-15% bottom mechanical stop of throttle plate, driver still open throttle via mechanical steel cable like on normal car, only bottom endstop is controlled via ecu.

controll mechanism is same like DBW it has brushed dc motor and potentiometer for position sensor

i would love to canibalise DBW controll but safety features dont aloow me to do so because there is only one position potentiometer inside idle mechanism, second potentiometer inside throttlebody is sensing absolute throttle plate position and they dont corelate when driver press pedal

mck1117, 2022-08-12:
FYI, a newer "full" DBW throttle fits in the same spot as these weird throttles. It's not recommended to use these throttles as they're obscure and hard to control.

mmatousek89, 2022-08-12:
i cant use normal dbw because of racing rules, it has to be cable driven,  idle jack i like to use for idlle and antilag too

closed 2022-08-12 by rusefillc