rusefi/rusefi #4438: Something with cranking and engine phase detection

rusefillc, 2022-08-13:
@ElDominio please provide unhappy tune and logs

ElDominio, 2022-08-13:
Will do as soon as possible

ElDominio, 2022-08-14:
https://rusefi.com/online/view.php?log=964

At exactly 5.8 seconds, exhaust VVT jumps from 0 to 360

https://rusefi.com/online/view.php?log=965

At exactly2.4 seconds, you can see VVT go from 0 to -360, then at 3 second, from -360 to +360. Is this affecting sync phase or should not matter?

rusefillc, 2022-08-14:
Engine phase is driven only by vvt intake at the moment

bulb could be #669 which is still unresolved

overall let me go over those logs :)

ElDominio, 2022-08-14:
https://rusefi.com/online/view.php?log=966

On this log, at 6.4 seconds exhaust VVT jumps to 360, at 6.5 seconds it comes back down to zero, and at exactly that point the car seems quite happy and cranks over, working perfectly after that.

rusefillc, 2022-08-14:
exhaust VVT jumps could be consequese of camResyncCounter going up. camResyncCounter might be the real deal

[image]

rusefillc, 2022-11-21:
We had fixes. Are able to reproduce this issue now?

@mck1117 what's your take?

mck1117, 2022-11-21:
I think this is probably fixed, or at least wasn't duped when we did fix it

rusefillc, 2022-11-21:
@ElDominio any objections to me closing?

ElDominio, 2022-12-03:
Close

closed 2022-12-03 by rusefillc