rusefi/rusefi #4439: Hellen NB1 1999 Miata no tach output

green21135, 2022-08-13:
Rusefi Hellen NB1 ecu is fully functional except no tech output to the gauge. Tach works in tuner studio properly but no signal to the dash. Tried setting different output pins in TS, does anyone know what pin/settings I’d need? Or is this a hardware issue

BeerMoneyMotorsports, 2022-08-14:
Tach output is fully functional. 

[image]

green21135, 2022-08-14:
> Tach output is fully functional.
> 
> [image]

I am getting this list of pins available, what do I need to do for it to list that pin?
[image]

rusefillc, 2022-08-14:
@green21135 how did you get fgis unit when?

Please update the firmware toblatest release

rusefillc, 2022-10-03:
Closing since no response

closed 2022-10-03 by rusefillc