rusefi/rusefi #4448: something wrong with enum to string build

mck1117, 2022-08-15:
[image]

rusefillc, 2022-08-17:
@mck1117 please clarify steps to reproduce

https://github.com/rusefi/rusefi/commit/f6fcd7387be288bbe38ae13b19eab08c695b26bb seems happy

mck1117, 2022-08-17:
those are all the steps I have - it does it for me on `master`

Relatively normal linux pc. Checked in binary works fine. ConfigDefinition.jar built with the same command works fine.

[image]

rusefillc, 2022-08-17:
So this is about :config_definition:shadowJar producing an invalid enum...jar for whatever reason.

:config_definition:shadowJar was not intended to even touch enum...jar in the first place