rusefi/rusefi #4451: invalid ini

mck1117, 2022-08-17:
[image]

rusefillc, 2022-08-17:
please elaborate

what version of TS? new issue? old issue? any context?

mck1117, 2022-08-17:
ini from current master, latest non-beta TS

mck1117, 2022-08-17:
broken by this commit: https://github.com/rusefi/rusefi/commit/afc376bf8c31bae118cf429fa2b4b576c45374e1

because when you said

[image]

that was in fact not quite true, because the legacy parser doesn't work either

mck1117, 2022-08-17:
This issue and the discussed `vvtStatus` "array of struct iterate" are both fixed by #4287.

rusefillc, 2022-08-17:
@mck1117 please clarify your test process for #4385

did you open TS after you broke .ini under ##4385?

commit 041c68ff (2022-08-17): invalid ini fix #4451
closed 2022-08-17 by rusefillc