rusefi/rusefi #4454: .ini files should be uploaded prior to snapshot upload

rusefillc, 2022-08-17:
At the moment build-firmware.yaml uploads snapshot potentially before corresponding .ini file is uploaded

We should move .ini file upload into firmware job

[image]

rusefillc, 2022-08-17:
Another benefit of moving .ini files to firmware job would be not uploading .ini file if we "skip-rate" firmware build. That's important since current REO host does limit total number of files.

mck1117, 2022-08-17:
> Another benefit of moving .ini files to firmware job would be not uploading .ini file if we "skip-rate" firmware build. That's important since current REO host does limit total number of files.

that's a good point! what's that file number limit?

rusefillc, 2022-08-17:
[image]

mck1117, 2022-08-17:
it also looks like we might be uploading every ini file for every build, even if they're duplicates

mck1117, 2022-08-17:
for example this build uploaded ini files with no ini change: https://github.com/rusefi/rusefi/runs/7829688078?check_suite_focus=true

rusefillc, 2022-08-18:
@mck1117 two issues

1) FullServerTest relies on default .ini file to be available in the internet
2) we upload custom builds but not the default .ini see https://rusefi.com/online/ini/rusefi/2022/08/

rusefillc, 2022-08-18:
@mck1117 

[image]

closed 2022-08-18 by rusefillc