rusefi/rusefi #4459: 350Z VQ35 tentative trigger issues

ElDominio, 2022-08-19:
https://rusefi.com/online/view.php?log=972

At 4.3s, RPM jumps to 900 even though the real engine did not do this, then at 6.35s goes from 300rpm to 150, which is exactly half of reality. Once it is somewhat stable the car starts and never has issues again

mck1117, 2022-08-19:
This is due to a limitation in how we measure RPM. We have an "alternative" way of doing it too called "instant rpm" that looks much better in your log. #4460 created.

rusefillc, 2022-08-19:
@ElDominio can you please elaborate? At the moment i do not understand your update

ElDominio, 2022-08-19:
In the log it shows the RPM jumping wildly on cranking, even though cranking is stable at 300 RPM.

Also in WOT pulls, RPM is jumping to 11k ish
https://rusefi.com/online/view.php?log=973

ElDominio, 2022-08-19:
https://rusefi.com/online/view.php?log=974

Lowered dwell by .2, problem is quite mitigated, pending further testing

mck1117, 2022-08-19:
Lots of weird things are going on.

Why is your MAP 120kPa? Turbo but only 3psi?

Why is your accelerator pedal dropping out during the pull?

Those RPM doublings are on trigger sync losses.

ElDominio, 2022-08-19:
Centrifugal supercharger
Acc pedal? no idea i might be scared mid pull, i'm also not revving out to max rpm since i usually do lower rpm limits until im comfortable with how the car behaves

ElDominio, 2022-08-19:
https://rusefi.com/online/view.php?log=976

VVT intake #1 can be seen jumping about 4 degrees (from 0 to 4) in idle, looks like a step function. Might be of interest?

ElDominio, 2022-08-19:
https://rusefi.com/online/view.php?log=977

This has a few more

rusefillc, 2022-08-19:
looks like crank decoding issue, everything else is a consequnce of crank decoding issue

[image]

ElDominio, 2022-08-19:
Here is a max data rate log at WOT-ish
https://rusefi.com/online/view.php?log=979

at 1.96s the issue is slightly apparent, at 3.9s it is very apparent

ElDominio, 2022-08-19:
[image]

ElDominio, 2022-08-19:
FOcus on RPM tripling

rusefillc, 2022-08-19:
https://github.com/rusefi/hellen-gm-e67/issues/51

rusefillc, 2022-08-19:
Please try flipping "invert primary" maybe it's a lazy hall issue?

[image]

ElDominio, 2022-08-19:
NOTE: Setting "invert primary" to "true" makes intake vvt jitter about 25 degrees at idle, log attached
https://rusefi.com/online/view.php?log=980

This log had a misfire around 3700-4000RPM
https://rusefi.com/online/view.php?log=981

rusefillc, 2022-08-19:
Please forget about VVT - we are trying to get reliable RPM without trippling. 

"misfire" could be a looooot of different issues

C202
[image]

mck1117, 2022-08-19:
cam is OK, crank is maybe not OK.

@ElDominio can you take a tunerstudio trigger log of cranking/starting? Your high RPM problem is also the primary wheel losing sync, nothing to do with cam.

ElDominio, 2022-08-19:
https://rusefi.com/online/view.php?log=983 
Cranking with "Invert primary" as false

https://rusefi.com/online/view.php?log=984
Cranking with "invert primary" as true

mck1117, 2022-08-19:
trigger log (the csv one), not a normal log

rusefillc, 2022-08-19:
hmmm why RPM loss

[image]

rusefillc, 2022-08-19:
side note? both files have very odd pattern of RPM in the beginning

[image]

mck1117, 2022-08-19:
@rusefillc that's partially due to vvt resync logic - the engine cycle changes length when we resync, so the next rpm calculation is weird

rusefillc, 2022-08-19:
that probably explains one of the dips and we have two.

ElDominio, 2022-08-19:
@mck1117 
[triggerlog.csv](https://github.com/rusefi/rusefi/files/9385560/triggerlog.csv)

I can't get any moer logs today but yeah, I'll try to come back to this car on monday

mck1117, 2022-08-19:
Will have a look this evening

mck1117, 2022-08-20:
These logs look fine, but they're at sensible RPM running. Possible to grab one of crank/start? Push start button in TS, crank/start engine, push stop button.

ElDominio, 2022-08-20:
This post has cranking logs https://github.com/rusefi/rusefi/issues/4459#issuecomment-1221101477 with injection disabled

mck1117, 2022-08-20:
but not trigger csv

mck1117, 2022-08-20:
> that probably explains one of the dips and we have two.

Unless there is a vvt mis-sync happening somewhere that resets the state, or a crank mis-sync that means we get reset twice.

mck1117, 2022-08-20:
> @rusefillc that's partially due to vvt resync logic - the engine cycle changes length when we resync, so the next rpm calculation is weird

Ooh, now I understand both drops.

[image]

Number 1 is a vvt resync changing the length of the engine cycle, tricking the RPM logic about when things happened. #4463

Number 2 is the transition from instant to not-instant RPM. If the engine accelerates rapidly (like during cranking, starting, etc), non-instant RPM may be lower, since the first sync point was long ago when the engine was turning slowly. The engine has since sped up, but it still gives the average RPM since the last sync point. Since the first sync point was at very low engine RPM, that was a long time ago, so the RPM calculated is really low. #4460 will fix this part.

ElDominio, 2022-08-20:
@mck1117 https://github.com/rusefi/rusefi/issues/4459#issuecomment-1221119530 this comment has trigger logger CSV's

mck1117, 2022-08-20:
try gap ratios like these, it can't hurt

[image]

rusefillc, 2022-08-21:
By the way we do have VQ logic analyzer files at https://github.com/rusefi/rusefi/tree/master/unit_tests/tests/trigger/resources

ElDominio, 2022-08-22:
Will be trying everything out tomorrow. The shop told me if I can't fix it tomorrow they will switch to LinkECU, I'll try to convince them otherwise

ElDominio, 2022-08-22:
I'm with the car, implemented the trigger gaps and updated the ECU, Trigger Angle seems to have changed, will document with new trigger angle as soon as we get the timing light on it

ElDominio, 2022-08-22:
[image]

Changed cam angles to these values and seems to work fine

ElDominio, 2022-08-22:
https://rusefi.com/online/view.php?log=990

Looks like no trigger loss, still some issues when getting on WOT but I'll have to dive deeper, does not look like sync loss to me, will report back if adding boost causes any issue

ElDominio, 2022-08-22:
I think this issue can be closed, new issue will be opened concerning something else

ElDominio, 2022-08-22:
https://rusefi.com/online/view.php?log=993

Triple RPM at about 5.5k

rusefillc, 2022-08-22:
https://github.com/rusefi/rusefi/issues/4469

mck1117, 2022-08-22:
current guess is that we're seeing a tooth get missed from over-eager verbose printing

rusefillc, 2022-08-22:
Do not we have a mechanism to shut verbose output above magic RPM?

mck1117, 2022-08-22:
[image]

mck1117, 2022-08-22:
This log looks like it loses crank sync first, then cam resyncs it (as it should). 

[image]

rusefillc, 2022-08-22:
5500 is above 2500 so no verbose hopefully?

mck1117, 2022-08-22:
see screenshot at 2681 rpm printing with threshold at 2500

rusefillc, 2022-08-22:
here verbose is consequence of trigger error condition

```
			bool verbose = engine->isEngineSnifferEnabled && triggerConfiguration.VerboseTriggerSynchDetails;

			if (verbose || (someSortOfTriggerError() && !silentTriggerError)) {
```

mck1117, 2022-08-22:
1. trigger error causes print
2. print causes subsequent event to get missed
3. missed event causes trigger error
4. GOTO 1

Should `silentTriggerError` be an `||` instead of `&&`? There is currently NO way to disable printing for lower tooth count wheels.

As an aside, this actually IS a "60 tooth" wheel, but it's implemented as a 20-edge repeated 3 times.

rusefillc, 2022-08-22:
@mck1117 I need your help issue tracking, I am failing to fork new issues for all separate-enough issues

rusefillc, 2022-08-24:
:(

I've just poked a recent eBay 121nissan customer to check his status

pk386, 2022-08-26:
@ElDominio @rusefillc 
I should be able to confirm with the 2005 Altima VQ35DE if you give me directions.

rusefillc, 2022-08-26:
@pk386 i believe it's about 5K revving with load, so you need a somewhat tuned instance. And a dyno ideally?

ElDominio, 2024-02-23:
So something is up with 350Z trigger especially in cranking, once it's running it's fine, bumping to keep tabs to get real life data

rusefillc, 2024-02-23:
Just had a similar story with subaru - gaps were widened to get more reliable craning while same as long as it runs it was happy

ElDominio, 2025-03-26:
Bad fucking sensors fuck nissan

closed 2025-03-26 by ElDominio