rusefi/rusefi #4463: Incorrect instant RPM immediately after VVT resync

mck1117, 2022-08-20:
Seen in this log: https://rusefi.com/online/view.php?log=983

[image]

Consider the event marked `1`.  At that instant, a vvt resync happened, resolving ambiguous crank position using cam information. This produced a discontinuity in the engine phase, which means we start comparing against the wrong teeth in the past to compute instant RPM. In this case, the "true" RPM at the skip was ~180, but since we started looking at the wrong spot in the trigger cycle to compute instant RPM, we came up with only 45 rpm, until the engine had come around again and written values in the correct positions.

ElDominio, 2022-08-20:
Somewhat related question: could this be happening to other triggers with VVT?

rusefillc, 2022-08-20:
short answer: it's complicated
this nissan is a very special case "tripple symmetrical, 120 degrees symmetrical"

Miata NB2 is "double symmetrical, 180 degrees symmetrical"

so it's not exactly the same as say hyuindai 60/2

mck1117, 2022-08-21:
@rusefillc it can happy with any crank pattern resolved by cam position, including 60-2 crank plus cam

mck1117, 2022-08-21:
remember a 360 degree crank pattern is still ambiguous, not just symmetrical crank options

mck1117, 2022-09-07:
Ah hah, looks like #4544 probably worked - that little flat spot with the blue arrow is the portion after reset where we simply don't calculate RPM for ~90 degrees until we get more data, avoiding the weird blip.

[image]

ElDominio, 2022-09-09:
https://rusefi.com/online/view.php?log=1037

Seems perfect now, I love you

closed 2022-09-09 by rusefillc