rusefi/rusefi #4466: Hyundai154 not showing model in firmware, is this ok?

ElDominio, 2022-08-22:
[image]

Used console to update, and now it seems to be running generic firmware. ECU works fine, but just a heads up

rusefillc, 2022-08-22:
[image]
 looks good

do you expect to see hyuindai somewhere else?

i.e. that's a big screenshot, what part of the screenshot looks wrong or what _exactly_ is not working as intended?

ElDominio, 2022-08-22:
That is the one it is expecting if i'm not wrong, the one found being 
[image]

rusefillc, 2022-08-22:
@ElDominio it looks like it was always like that (console title never showed board name) and nothing is "broken" but it's overall "confusing"

```
	char versionBuffer[32];
	chsnprintf(versionBuffer, sizeof(versionBuffer), "rusEFI v%d@%s", getRusEfiVersion(), VCS_VERSION);
	tsChannel->sendResponse(TS_CRC, (const uint8_t *) versionBuffer, strlen(versionBuffer) + 1);

```

mck1117, 2022-08-22:
this looks fine....does the "yes" button work?

rusefillc, 2022-08-22:
Yes but also legit usability improvement suggestion or at least signal, it's fair to be confused by current inconsistency

ElDominio, 2024-02-23:
something was changed about naming on firmware and whatever is the current state is fine

closed 2024-02-23 by ElDominio