rusefi/rusefi #4470: Damn what is "useOnlyRisingEdgeForTriggerTemp" from 2018?

rusefillc, 2022-08-22:
```
	bool useOnlyRisingEdgeForTrigger = triggerConfig.UseOnlyRisingEdgeForTrigger;
	this->useOnlyRisingEdgeForTriggerTemp = useOnlyRisingEdgeForTrigger;
```

"temp" implies that something was supposed to get better _later_?

mck1117, 2022-08-22:
There do seem to be two ways to do "use only rising":

- the trigger can set it (`useRiseEdge`)
- the user can set it (`useOnlyRisingEdgeForTrigger`)

and they seem to do subtly different things!

rusefillc, 2022-08-22:
right, but what is proper name for "useOnlyRisingEdgeForTriggerTemp" class member? Is that member needed as a separate member?

mck1117, 2022-08-22:
I have no idea. It might be worth a refactor here anyway, is my point.

Do we ever need the user to set "use only rise"? Can we force that on except for the few triggers that need it?