rusefi/rusefi #4474: Set a complete fuel cut when BattV = 0v

ElDominio, 2022-08-23:
This would make cars with main relay control shut off the moment key is shut off, then turn off the main relay after 1sec as usual. Clients keep asking why their car stays on after they key off

rusefillc, 2022-08-23:
Let's unpack if multiple action items here:

1) is this happening with USB power or without or both
2) why 1 second for main relay what is it about

mck1117, 2022-08-23:
besides, we should already be cutting the main relay when the ignition voltage goes low

ElDominio, 2022-08-23:
1) happening on both
2) some cars get angry if it all shuts down at once? idk lol but if we want to keep MR alive for any reason, cutting fuel should still be done even if MR is on

shadowm60, 2022-08-23:
On motorcycles there is an issue sometimes that voltage regulator dies, this would make voltage to jump over 18v, maybe this issue could be broken down that main relay/injectors are fired if the voltage is in a preconfigured voltage range?

rusefillc, 2022-08-23:
So, overall: we have no idea what's going on and we have a lot of ideas :)

let's start the game of elimintation: @BeerMoneyMotorsports @stefanst did you experience anything relevant on Miata NB2?

BeerMoneyMotorsports, 2022-08-23:
No issues with my NB2 or any  reported by my customers..

rusefillc, 2022-08-23:
@ElDominio it sounds like this issue is only specific to specific vehicles.

Question1: are you aware of vehicles where we do NOT have the issue?
Question2: any model/year where we DO have the issue? Wiring diagrams etc?

ElDominio, 2022-08-23:
Hellen154hyundai

69 mustang with a 4chan controlling main relay

mck1117, 2022-08-23:
@ElDominio can you draw a rough diagram of:

- ignition wire from key to ECU (and anything else relevant)
- main relay wiring to/from ECU
- main relay wiring to other engine hardware

rusefillc, 2022-08-23:
@mck1117 see edit - looks like Hellen154hyundai  is the one to explore? https://github.com/rusefi/rusefi_documentation/tree/master/OEM-Docs/Hyundai

mck1117, 2022-08-23:
does the behavior still happen with USB connected? ie, can you log it?

rusefillc, 2022-09-28:
@mck1117 what shall we do in your opinion? close or ask a question?

ElDominio, 2022-09-28:
Remind me and I'll get a video of it happening

mck1117, 2022-09-28:
After changing some wiring on my race car specifically in the name of increased safety during faults (if you crash, fuel pump should stop!), I think it's worthwhile doing this anyway. What if the car is wired wrong and the engine stays on? What if the main relay welds itself shut? We should kill the engine!

rusefillc, 2022-10-04:
@ElDominio has provided https://youtu.be/qjGH8XD5hUM

mck1117, 2022-10-04:
If the laptop is connected in that vid, can we get a TS log of that happening? I'm really curious what is going on there.

ElDominio, 2022-10-05:
Car keeps 12v input at 12v for a second because thanks Mitsu

It still behaves strangely on hellen154 but, not like the video.

mck1117, 2022-10-05:
If the ignition pin keeps voltage on it, how would we know that it's turned off?

ElDominio, 2022-10-05:
That's why I didn't upload a log of it, on that car we don't have much of a choice. Once a Mustang I have at the shop I'll see if I can replicate what I mean, but on the Evo we can't do diddly squat, maybe I have to sniff key state from CAN (probably)

ElDominio, 2022-12-03:
Close

closed 2022-12-03 by rusefillc