rusefi/rusefi #4476: windows 7 test log

rusefillc, 2022-08-23:
i have a win 7 laptop, just hooked MRE and no connectivity

[image]

rusefillc, 2022-08-23:
I've installed drivers using console button and got port working

[image]

rusefillc, 2022-08-23:
switched to DFU using console but unable to DFU update

[image]

mck1117, 2022-08-23:
do we care about w7?

rusefillc, 2022-08-23:
I have zero attachement to win7.

#4475 would at least give us a chance to know what our users are using

In my specific case win7 is the device I can just grab and test with zero effort upfront. With device drivers I assume that we would need real HW test, I am not aware of any fancy virtualization to help us here? Looks like _someone_ would have to use a bunch of hard drives for different Windows versions?

rusefillc, 2022-08-24:
```
::echo off

@echo off
setlocal
for /f "tokens=4-5 delims=. " %%i in ('ver') do set VERSION=%%i.%%j

if "%version%" == "6.1" (
if exist "%windir%\sysnative\pnputil.exe" (
    start "STM32 USB DFU DRIVER" %windir%\sysnative\pnputil.exe -i -a  %0\..\DriverNotSigned\DFU_in_HS_Mode.inf
) else (
    start "STM32 USB DFU DRIVER" pnputil -i -a  %0\..\DriverNotSigned\DFU_in_HS_Mode.inf
)
) else (
if exist "%windir%\sysnative\pnputil.exe" (
    start "STM32 USB DFU DRIVER" %windir%\sysnative\pnputil.exe -i -a  %0\..\Driver\STM32Bootloader.inf
) else (
    start "STM32 USB DFU DRIVER" pnputil -i -a  %0\..\Driver\STM32Bootloader.inf
)
)
endlocal
```

rusefillc, 2022-08-24:
https://stackoverflow.com/questions/7044985/how-can-i-auto-elevate-my-batch-file-so-that-it-requests-from-uac-administrator

closed 2022-08-24 by rusefillc