rusefi/rusefi #4477: ECU does not fire inj/ign correctly after swapping settings

ElDominio, 2022-08-23:
I use one 4chan ECU for:
1) Honda B20 (running a modified 24-1 modified trigger)
2) Mitsubishi 3A92 3 cylinder

When I moved the ECU from 3A92 to B20, injector 4 did not fire until i updated firmware (dk what this has to do with it)

Then, I swapped back to 3A92,  and now coil 3 does not fire (shows on RusEFI console that Coil3 is not being commanded to fire)

Logs and tune at
https://rusefi.com/online/view.php?log=996

mck1117, 2022-08-23:
snip of rusefi console?

please include all the information you can about an issue

ElDominio, 2022-08-23:
[image]

mck1117, 2022-08-23:
I don't understand your trigger configuration

[image]

[image]

Does this engine have vvt? Even if not, the "first" cam must always be in the spot labeled "intake cam". But you have no cam inputs configured at all?

mck1117, 2022-08-23:
There is a real bug here though, we're trying to do wasted spark on an odd-cylinder engine which doesn't really work properly right now.

mck1117, 2022-08-23:
For now you should set "Require cam/VVT sync for ignition" to `true`

ElDominio, 2022-08-23:
It does have VVT, and it is only on the exhaust cam, and only has that cam sensor. Intake has no sensor.

It did run correctly before with these settings, quite well in fact https://youtu.be/XcJ1Yui71R4 haha

I'll swap those settings and report back once I can (just stored the car)

mck1117, 2022-08-23:
Ok, well, the cam input set as secondary is never correct for this setup.

mck1117, 2022-08-23:
see also #4195

ElDominio, 2022-08-24:
Will try to test this on Thursday

ElDominio, 2022-09-01:
Not working, now it doesn't start since both coils 1 and 3 fire at the same time

log and tune: https://rusefi.com/online/view.php?log=1019

console snip: 
[image]

ElDominio, 2022-09-01:
[image]

Set to sequential ignition

ElDominio, 2022-09-01:
[image]

Injection enabled

mck1117, 2022-09-01:
This one looks kind of similar: #4421

rusefillc, 2022-09-05:
Related to #4532 - need to a recording of that vehicle

We did not change anything relevant for a while, this must be a tune issue. Worst case try a couple of previous releases.

https://rusefi.com/online/view.php?msq=1201&dialog=triggerConfiguration_IO uses proper cam option

Why wasted spark on https://rusefi.com/online/view.php?msq=1201&dialog=ignitionSettings ? you have CAM please use it for sequential?

rusefillc, 2022-09-05:
Actually #4533 was changed 12 days ago so maybe do try Aug release just to confirm

but wasted does not right here, you have CAM please use it to reduce complexity

ElDominio, 2022-09-05:
https://github.com/rusefi/rusefi/issues/4477#issuecomment-1234811305

This was done with sequential ignition/injection.

rusefillc, 2022-09-05:
I would love to be effective
@ElDominio please join me in my desire to be effective

If we mix and jump between tunes need full tunes.

I suspect that the signal here is "tools such please implement better tools". It sounds like console should compare latest online tune to current tune and offer convinience, or maybe I need to fix TS plugin.

ElDominio, 2022-09-05:
but the full tune is in online, is it not?

rusefillc, 2022-09-05:
@ElDominio a tune not explicitly linked pretty much does not exist

@ElDominio a tune not explicitly linked means "please find me and good luck hopefully you would find the right tune but maybe not"

@ElDominio please link the tune you are referring to

ElDominio, 2022-09-05:
https://rusefi.com/online/view.php?log=1029

ElDominio, 2022-09-06:
Andrey says ignition cut reason = 12 which is "no engine phase"

mck1117, 2022-09-06:
[image]

rusefillc, 2022-09-06:
expected https://github.com/rusefi/rusefi/wiki/Images/triggers/trigger_TT_VVT_MITSUBISHI_3A92.png

mck1117, 2022-09-06:
12 = engine phase (cam didn't sync yet

2 = fuel (aka the "enable fuel" bit is disabled on the injection config page)

mck1117, 2022-09-06:
> expected https://github.com/rusefi/rusefi/wiki/Images/triggers/trigger_TT_VVT_MITSUBISHI_3A92.png

trigger looks great IMO, look at nice instant RPM (except for the blip that we already know about #4463)

ElDominio, 2022-09-06:
[image]

https://rusefi.com/online/view.php?log=1033

mck1117, 2022-09-06:
Your tune loaded in simulator looks completely normal:

[image]

rusefillc, 2022-09-06:
Huh? Is not simulator supposed to NOT work sice not VVT sync?

mck1117, 2022-09-06:
I have cheated a little bit and tricked the sim in to having vvt sync

ElDominio, 2022-09-06:
[rusEFI_trigger_log_2022-09-06_18_10_08_099.zip](https://github.com/rusefi/rusefi/files/9501180/rusEFI_trigger_log_2022-09-06_18_10_08_099.zip)

ElDominio, 2022-09-06:
[image]

mck1117, 2022-09-06:
yooo what did you change?

rusefillc, 2022-09-06:
@mck1117 just edited with more info

closed 2022-09-23 by rusefillc