rusefi/rusefi #4500: AC Compressor Delay Option

10AE0739, 2022-08-30:
For tuning idle with AC, I think it would be helpful to have a programmable delay between increasing the target RPM and the AC compressor coming on. I’m currently working on (and learning) tuning, but my RPM dips from the additional load before the RPM has had time to increase by 200. I assume a delay in milliseconds (500 or 1000) could help with this.

mck1117, 2022-08-30:
Do you have closed loop idle timing enabled, and far enough retarded idle timing for it to be useful?

10AE0739, 2022-08-30:
Yes. I tried to create a log during lunch to upload but I get an error message on upload and it's only capturing the first part of the log, which doesn't include any of the AC cycling on and off. Maybe I can get a working log after work. Current tune is uploaded.

10AE0739, 2022-08-31:
I've uploaded a couple logs as well.

mck1117, 2022-08-31:
@10AE0739 feature has been added to latest snapshot (will build within an hour from now), give it a try?

closed 2022-08-31 by mck1117

10AE0739, 2022-09-01:
I'll update and try that soon. Thank you!

10AE0739, 2022-09-01:
I'm working on testing right now. Should be unrelated, but CAN Wideband is no longer reading in Tuner Studio.