rusefi/rusefi #4501: CAN Calibration Protocol

rusefillc, 2022-08-30:
https://github.com/pySART/cccp is small

rusefillc, 2022-08-30:
https://www.csselectronics.com/pages/ccp-xcp-on-can-bus-calibration-protocol

```
2022-08-30_14_55_04_594: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 00 34 12 00 00 00 00
2022-08-30_14_55_04_785: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 01 34 12 00 00 00 00
2022-08-30_14_55_04_988: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 02 34 12 00 00 00 00
2022-08-30_14_55_05_192: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 03 34 12 00 00 00 00
2022-08-30_14_55_05_500: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 04 34 12 00 00 00 00
2022-08-30_14_55_05_695: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 05 34 12 00 00 00 00
2022-08-30_14_55_05_894: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 06 34 12 00 00 00 00
2022-08-30_14_55_06_094: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 07 34 12 00 00 00 00
2022-08-30_14_55_06_405: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 08 34 12 00 00 00 00
2022-08-30_14_55_06_595: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 09 34 12 00 00 00 00
2022-08-30_14_55_06_798: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 0A 34 12 00 00 00 00
2022-08-30_14_55_07_000: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 0B 34 12 00 00 00 00
2022-08-30_14_55_07_296: EngineState: CAN RX bus 0 ID 7EF(2031) DLC 8: 01 0C 34 12 00 00 00 00

```

[image]

rusefillc, 2022-08-31:
https://github.com/christoph2/pyxcp/issues/114

rusefillc, 2022-08-31:
https://rusefi.com/forum/viewtopic.php?f=5&t=2419

rusefillc, 2022-08-31:
https://www.ni.com/docs/en-US/bundle/ecu-measurement-calibration-toolkit/page/niecumchelp/xcp_spy.html

[image]

rusefillc, 2022-11-13:
https://www.kvaser.com/about-can/higher-layer-protocols/ccpxcp/

rusefillc, 2022-11-13:
@rnd-ash do you know anything about XCP/CCP? Any ECU I can get on eBay with seedkey and XCP/CCP to see this in action?

rusefillc, 2022-11-13:
@CanBusHack any ideas by any chance?

rnd-ash, 2022-11-13:
I've heard of this briefly  before, but never really touched  on it or looked into it. From what I remember looking at, this protocol is only really used during ECU development, so I don't think there is any ECU out there commercially that has this enabled, as its likely disabled on production software. That said, it may be worth trying to get your hands on development ECUs somehow...

rusefillc, 2022-11-13:
@rnd-ash here's a totally random MED17 from Nick's collection

Note how ASAP2Demo which is an XCP/CCP software says a happy "connected" and drama seems to be about seedkey not protocol in general. This supports a statement that many used normal ECUs have the protocol on board it's just hidden under seedkey.

[image]