rusefi/rusefi #4511: tear down Engine god object

rusefillc, 2022-09-01:
God object not good for too many reasons. 

At the moment I am mostly annoyed by "any change of any header causes full recompilation which is not instantanious even on WSL"

While the most annoying thing is full recompilation, there is also the challenge of unit testing and incapsulation. At the moment our design is extremely questionable and having explicit control over dependencies would be nice. At the moment we do not see cyclic dependencies etc etc.

2627 is definitely not the most important one :)

rusefillc, 2022-09-01:
@mck1117 @racer-coder would you guys be interested to suggest a new pattern using any of the smallest controllers as an example?

mck1117, 2022-09-01:
> there is also the challenge of unit testing and incapsulation

this doesn't necessarily break encapsulation - see various unit tests that test individual engine modules (not touching the ones inside the `Engine` object)

> While the most annoying thing is full recompilation

This is also (mostly?) an effect of the precompiled header, not that we have a stack of engine parts cooked in to a class called `Engine`.

mck1117, 2022-09-01:
I do agree that god objects aren't great, but all of those parts have to sit _somewhere_.

There are two slightly competing interests:

1. be able to initialize/destroy the whole world in a predictable way in a unit test. Right now that means creating/destroying an instance of `Engine`.
2. have minimal coupling between various parts

In order for #1 to be possible, we have to put some "bag of modules" somewhere, so that a unit test doesn't have to `new` up 57 different objects.  Today, the logic and state are clearly broken in to the responsible modules, with most modules having their own standalone tests that don't rely on the `Engine` itself.

Do we think this is actively causing a problem, right now, today? It seems like re-architecting this a second time in under a year is maybe needless churn in search of a problem.

rusefillc, 2022-09-01:
Yes, absolutely - unit test full initialization is one of the key elements. We just need a pattern/one shared method for "full" reinit, and an option to more manually assemble subset of objects more manually. 

> Do we think this is actively causing a problem, right now, today? It seems like re-architecting this a second time in under a year is maybe needless churn in search of a problem.

Yes. I am failing to comprehent current fuel scheduling - I would like to refactor fuel scheduling into readable state. I would also like to break fuel scheduing into proper modules. The simplest way seems to be being able to see dependencies explicitly, not via ``Engine``

racer-coder, 2022-09-01:
You're compiling for a machine that only has a meg or two of flash.  Modularization will only buy you so much.
I kinda tried going down this path with the EngineModule concept - the idea that you can have a bag of goodies and reference the one(s) you need.  You could have separate bags of goodies for different unit tests in order to prove less dependence, but then you need to templatize all the users of the bag so that the bag itself is a template parameter, which means moving a bunch of code from .cpp files to .h files, which means longer compile times.  So that's counter productive.

If it's compile time you're after, precompiled headers are a blessing and a curse at the same time, as @mck1117 said.
If it's link time you're after, maybe look at libraries (maybe dynamic ones?) so less linking needs to happen when small changes are made, at least for unit tests?  Disable LTO, etc.  The downside is you really want unit tests to be as close to the actual product as possible so that you're testing what you're shipping, which means same compiler, same linker, etc.  Or you make sure github tests both.  Otherwise, clang and it's linker are faster than gcc/ld, but they usually generate worse code in my experience.

mck1117, 2022-09-01:
> You're compiling for a machine that only has a meg or two of flash.

Not to mention that it's a few hundred K of RAM and no memory paging at all

rusefillc, 2022-09-01:
I have created this ticket today because I am failing to comprehend fuel scheduling layer - too many objects with potentially circular dependencies. My primary problem is comprehending the code flow in the codebase.

rusefillc, 2022-09-01:
#1594 is begging for this area of the codebase to become comprehendable

mck1117, 2022-09-01:
> Otherwise, clang and it's linker are faster than gcc/ld

testing on my machine (WSL on windows 11, code checkout in WSL filesystem not NTFS), running `make -j12` (clang does build and pass tests :
clang: 3m4s
gcc: 1m26s

Not faster, at least with our current set of options.

racer-coder, 2022-09-01:
Clang precompiled header options are not compatible with gcc.  I remember having to play around with this last time I looked.

racer-coder, 2022-09-01:
ok if you want to go C++ crazy (should I just stop now?)
Make a derivation from EngineModule that takes a templated list of modules that your module is allowed to access.  Make a constructor that takes the master list as an input.  Make an accessor function, like Engine::module() to access modules.  Validate that any module accessed via the function exists in the list of modules you expect to access.

Change permissions on Engine::module() so that only this new derivation can actually access it. 

i.e.
template<typename all_my_modules_i_can_use...>
struct module_accessor : EngineModule {
  template<typename get_t> auto & module () {
    static_assert(get_t is in all_my_modules_i_can_use);
    return Engine->module<get_t>();
  }
};

change in Engine

private: template<typename get_t> auto & module();
template<typename stuff...>
friend class module_accessor<stuff...>;

If you want to force non-cyclic dependencies, then force everything in engineModules to be ordered by use, and change module_accessor to make sure the stuff you're looking up comes before child (so you'd have to add typename child_t to module_accessor template parameter list).

Whether any of this is worth it is an exercise left to the reader.  Caution, there is no right answer.

mck1117, 2022-09-02:
> Caution, there is no right answer.

Agree, perfect is absolutely the enemy of good. The current solution seems at least OK.

> force non-cyclic dependencies

I think cyclic dependencies are either invalid, nor should we prevent them. Consider AC and idle. You might need them to both deal with the other one. Turning on AC bumps idle speed up. Idle speed too low might kick AC off.

rusefillc, 2022-09-02:
Let's forget about cyclid dependencies for now.

@mck1117 do you want to attempt what @racer-coder is proposing? I am still a C++ virgin.

mck1117, 2022-09-02:
So, funny thing, I've flipped through our existing engine modules, and none directly reference another. There are no scenarios that we would call this new function.