rusefi/rusefi #4524: 12-3 trigger losing sync, ECU disables ignition and injection

ElDominio, 2022-09-03:
We installed a 2chan on a CRF250F (one cylinder, 12-3 crank trigger, using MAP to sync), and are having some strange sync issues. If the thing runs perfectly all the time, it'll never have any issues, but if there's a misfire or a sudden change in engine speed it seems to lose sync and disable injection/ignition, which is terrible since it's a bike (injection and ignition are disabled until RPMs are zero, even though it is reading RPM just fine). 

Logging the issue is a bit hard since on the dyno, it's fine (everything is under control)  Still, here is tune and some logs: https://rusefi.com/online/index.php?vehicleName=CRF&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

rusefillc, 2022-09-05:
Do we have any log of an incident?

ElDominio, 2022-09-05:
Very difficult to grab, behavior is a bit erratic. On dyno it barely happens, i'll have it back in the shop tuesday to try to reproduce

mck1117, 2022-09-05:
💀💀💀💀💀

[image]

rusefillc, 2022-09-05:
Lol 2022.03

ElDominio, 2022-09-05:
I always hit autoupdate and update firmware, wtf? 
Anywho it'll be at the shop tomorrow and I'll make sure it's on latest

ElDominio, 2022-09-11:
https://rusefi.com/online/view.php?log=1039

Updated today and it still reports as 2022.03.23, wtf?

ElDominio, 2022-09-11:
[image]

on latest snapshot

ElDominio, 2022-09-11:
[image]

ElDominio, 2022-09-11:
Fixed by using console commands for read out removal

https://rusefi.com/online/view.php?log=1040  <- primary inverted

https://rusefi.com/online/view.php?log=1041 <- primary not inverted

rusefillc, 2022-09-15:
@ElDominio please call I would like to empower you

[image]

https://rusefi.com/online/view.php?msq=1221&dialog=triggerConfiguration_gap

ElDominio, 2022-09-15:
[image]

rusefillc, 2022-09-15:
Damn I wish I knew we are tring the VTwin mode :)

Red line looks to be the same every 360 degrees right?

ElDominio, 2022-09-15:
[image]

No instant MAP!

ElDominio, 2022-09-16:
https://youtube.com/shorts/TsgtuGhcpm8?feature=share
https://rusefi.com/online/view.php?log=1044

Video and log of problem

problem not being a problem for reasons uknown
https://youtube.com/shorts/wiGEJNaRO6g?feature=share

rusefillc, 2022-09-16:
Data Rate: ~ 100
@ElDominio why are you so cruel to me?

Unfortunatelly even at 1KHz we do not have enough bandwidth to catch individual gap ratios. We would need to use external logic ananlyzer once you get latest 2chan in your hands with crank hookup point.

until we have logic ananlyzer pic i suggest starting randomly widening trigger gaps
try changing 4.7 to 5.7

try changing 0.800 to 0.7
try changing 1.340

[image]

rusefillc, 2022-10-03:
@ElDominio how is this going?

@mck1117 "reliabity was a confurn" I recall, changes changlogs issue tracking etc?

ElDominio, 2022-10-03:
Need logic analyzer? with new 2chan i should be able to

rusefillc, 2022-10-03:
4chan but yes. weird that it has just stopped to even start without any clear reason :(

ElDominio, 2022-10-03:
This 12-3 is on a 2Chan mounted on a CRF250, do the newer 2chan have pads?

mck1117, 2022-10-08:
time for logic analyzer, without that we can't do much here

ElDominio, 2022-10-08:
I think i can do this using a 4chan, will try asap

ElDominio, 2022-10-08:
[12minus3logclog.zip](https://github.com/rusefi/rusefi/files/9740272/12minus3logclog.zip)

Ran for a sec then hit the weird engine off condition

mck1117, 2022-10-10:
Your crank sensor lost a tooth.

[image]

mck1117, 2022-10-10:
and it did it again 5 revolutions later:

[image]

mck1117, 2022-10-10:
And just before it lost two teeth in the same rev:

[image]

mck1117, 2022-10-10:
Update: the VR threshold table was set too hot, reducing it up at those rpm should fix it.

mck1117, 2022-10-11:
User reports that it's fixed with adjusted VR threshold.

closed 2022-10-11 by mck1117

rusefillc, 2022-10-11:
https://github.com/rusefi/hellen88bmw-issues/issues/52