rusefi/rusefi #4529: All hard cuts (boost cut especially) should have hysterisis

ElDominio, 2022-09-05:
Client is riding the boost cut limit, making it activate/deactive very slightly on the top end. Example here:
https://rusefi.com/online/view.php?log=1028

ECU should disable injection for an amount (user configurable) of boost before enabling again, as to reduce confusion of misfire vs actual cut

mck1117, 2022-09-05:
yeah I hit the same thing the other day where I was getting very "buzzy" boost cut due to rapidly cycling on/off

mck1117, 2022-09-05:
TODO after #4541: do we care about configurable hysteresis, or is fixed fine?

ElDominio, 2022-09-05:
I always say configurable cause a 10kpa hysteresis on a car that make 150kpa total is not the same as 10kpa on a car that makes 300

ElDominio, 2023-02-02:
Sstill relevant

ElDominio, 2023-04-02:
STILL RELEVANT

ElDominio, 2023-04-08:
[image]

in area
https://rusefi.com/online/view.php?msq=1438&dialog=limitsAndFallback

[image]

A delay would be nice too but that's another separate ticket lol

rusefillc, 2023-07-11:
@ElDominio poke?