rusefi/rusefi #4538: Why is getsensor allowed to return nil?

rusefillc, 2022-09-05:
Say CLT as a thermister, should never return nil from common sense?

mck1117, 2022-09-05:
nope, if the sensor is failed it returns nil. If you unplug the CLT sensor, what should it return? 0? -1? Those are still valid CLT values. So instead we return nil, which means "no clt value available".

ElDominio, 2022-09-06:
Most ECUs return a known error value, like 
[image]

The error is logged and a DTC is thrown (to alert everything that CLT is in error), but a value is substituted so things that depend on it go to a safe operating parameter instead of having unhandled errors (like Lua currently is doing)