rusefi/rusefi #4539: VVT Cam position issues: Random jumps

ElDominio, 2022-09-05:
Lancer 4B11 has similar issues to Genesis (their cam setups are identical, half moon)

At random times, the cam reading jumps by an impossible value and then back to zero, at which point the car misfires for a split second

Log and tune: https://rusefi.com/online/view.php?log=1030

How it looks:
[image]

mck1117, 2022-09-05:
If you invert the vvt input (and adjust vvt offset to zero it out again), does that change the behavior? It's possible that one of the edges on the hall sensor is "lazy" and the other is "sharp".  We want the "sharp" one, not the "lazy" one.

ElDominio, 2022-09-06:
You are correct, switched the cam edge and now it works perfectly. Thanks @mck1117 !

ElDominio, 2022-09-07:
BTW, it fixed it for 4B11, NOT for Genesis! Genesis at cranking RPM is still unreliable sometimes

@rusefillc  automatic detection would be great, is it doable?
Maybe check for the variation in degrees in a certain amount of time, and pick the one with the least error? That's better than any other ECU out there

mck1117, 2022-09-07:
Not sure fully automatic is a very viable strategy. That's a great way to find some marginal condition (new combination of temperature/engine speed/whatever) that picks wrong and then your engine won't start/run properly.

> NOT for Genesis! Genesis at cranking RPM is still unreliable sometimes

log?

ElDominio, 2022-09-07:
https://rusefi.com/online/view.php?log=986 seen at 2s

https://rusefi.com/online/view.php?log=985

https://rusefi.com/online/view.php?log=964

Maybe have independent rising/falling for each sensor? idk why it still happens on genesis, and notice all logs show only exhaust cam

mck1117, 2022-09-07:
Those are all old firmware - log again with new snapshot on the genesis and see if behavior changed.

964 shows the crank losing sync

985 shows good vvt position, but shows #4463, which should now be fixed-ish. This log looks OK to me.

986 same as 985

These all look like different issues, some of which are fixed.

ElDominio, 2022-09-07:
https://rusefi.com/online/view.php?log=1035
Exhaust vvt on newest firmware

mck1117, 2022-09-07:
> https://rusefi.com/online/view.php?log=1035 Exhaust vvt on newest firmware

What happens when not cranking? We don't use that cam for any engine sync purpose, just vvt measurement.

mck1117, 2022-09-07:
OK, I now understand what's going on with exhaust vvt. Will create a new issue.

mck1117, 2022-09-08:
@ElDominio do a bunch of cranks back to back again on genesis, see how it looks now

ElDominio, 2022-09-12:
Forgot to update: VVT angles always look perfect, at least in this case! Need to test on 4B11 but G4KF is perfect!

ElDominio, 2022-12-03:
This can be closed as it is the same as 4b11 double rpm ticket

closed 2022-12-03 by rusefillc