rusefi/rusefi #4545: VVT minimum RPM

ElDominio, 2022-09-07:
VVT should have a minimum activation value, since at lower rpms (idle), targetting 0 could still result in the ECU slowly going crazy trying to control VVT position and outputting some duty when none is desired (Some car have a target of 0, slowly start misfiring, I set maximum duty to 0 and the misfire goes away).

ElDominio, 2022-10-07:
Issue with this: 
[image]

Max should be near 3k or so

rusefillc, 2022-10-07:
@ElDominio please claridy

original description focuces on idle " minimum activation RPM, since at lower rpms (idle)"

3000 rpm seems quite high for idle, is this a separate use case or else?

ElDominio, 2022-10-07:
Error says I can't set over 100rpm in picture

rusefillc, 2022-10-07:
@ElDominio is there anything you want to share

[image]

ElDominio, 2022-10-07:
I have a dealer in China

rusefillc, 2022-10-07:
fixed it's 3000 now

commit 974f6f54 (2022-09-28): VVT minimum RPM fix #4545
closed 2022-10-07 by rusefillc