rusefi/rusefi #4551: 2011 tahoe pedal calibdation

rusefillc, 2022-09-09:
we fail to calibrate pedal

https://rusefi.com/online/view.php?log=1036

https://rusefi.com/online/view.php?msq=1217

mck1117, 2022-09-09:
Looks like it should work fine.

[image]

More info about what "fail" means?

mck1117, 2022-09-09:
calibration in the tune even looks correct

rusefillc, 2022-09-09:
and yet PPS gauge is not happy - goes from 0% to 66%, see log

[image]

(random piece of info - there is a chance that one of pedal sensor inputs was changed without restart, should not matter?)

rusefillc, 2022-09-10:
and now it works. looks like some sort of reboot is needed to apply?

ideally we would need to break calibration and reproduce from scratch