rusefi/rusefi #4552: "Blip" with trigger error on several boards

BeerMoneyMotorsports, 2022-09-09:


rusefillc, 2022-09-09:
Vaguely related to https://github.com/rusefi/rusefi/issues/4553

BeerMoneyMotorsports, 2022-09-09:
i could not upload this file to REO but i think i had the issue at the end of this log.  ive been driving and logging all day.  it was at idle and just decided to shut off.  Not peter out and die, just cut off.  https://drive.google.com/file/d/1LgBXt_V72_rVc6iSb9b8lJ7vOliuQ3Fm/view?usp=sharing

rusefillc, 2022-09-09:
@BeerMoneyMotorsports please clarify which of four kinds of log files is there

@BeerMoneyMotorsports would you be available to paste a screenshot focusing on the relevant portion of the log here showing RPM and trigger error counter? that kind of raw data processng / more clear description of what you have would help.

rusefillc, 2022-09-09:
@mck1117 check this out we have a loss of second in time at the moment of trigger error

https://www.rusefi.com/online/view.php?log=1025 - note blue value jump from 662.6 to 663.6
[image]

[image]

rusefillc, 2022-09-09:
Same in 1022

This is highly suspecious :)

rusefillc, 2022-09-09:
Reminds us of #4554

BeerMoneyMotorsports, 2022-09-09:
It is a Tuner Studio log.  I have been logging all day in an attempt to catch it.  On my car, both occurrences show no trigger error. They were both around idle and then the engine cuts off.  

[image]
 
another occurrence https://rusefi.com/online/view.php?log=1038

[image]

BeerMoneyMotorsports, 2022-09-09:
@rusefillc both of mine have 1 second loss as well.

rusefillc, 2022-09-10:
https://github.com/rusefi/rusefi/issues/4557

BeerMoneyMotorsports, 2022-09-10:
@rusefillc i had several instances tonight on a date with the wife.  below is one of them and they all show 1 second missing.  These were all with SD logging at 3ms.

[image]

rusefillc, 2022-09-10:
The whole 1s jump is exciting progress. Now that I was able to reproduce this on a bench twice we are getting closer to resolution. https://github.com/rusefi/rusefi/issues/4557

BeerMoneyMotorsports, 2022-09-10:
@rusefillc i changed from 50ms to 3 ms on sd logging earlier. The issue happened far more frequently after that change.

rusefillc, 2022-09-10:
See also https://github.com/rusefi/rusefi/issues/4560

closed 2022-09-14 by rusefillc