rusefi/rusefi #4553: SD card data rate

rusefillc, 2022-09-09:
https://rusefi.com/online/view.php?log=1022 seems to be an example of a data log showing only 19Hz rate

SD Card Logger -> SD Card Logger default value is 50ms which means 1000 / 50 = 20Hz seems to align well!

rusefillc, 2022-09-09:
@BeerMoneyMotorsports since you drive can you try lower period maybe 10ms or 3ms?

@mck1117 do you still drive any rusEFI vehicles periodically?

rusefillc, 2022-09-14:
@BeerMoneyMotorsports poke did you have a chance to put any miles with lower period/higher frequency?

@mck1117 are you ever driving anything, what's your take on rate and period?

mck1117, 2022-09-14:
He's had it turned off for a bit since the hang bug.

I logged all weekend with SD at 20hz, didn't have a reason to go faster (also knowing that the bug exists). That said I'll probably turn it up to ~50hz now that the bug is fixed.

Do we think there is a code change that should happen? 20hz seems like a reasonable balance between "collect useful engine data" and "not too much storage" to me.

rusefillc, 2022-09-14:
The only code change would be to change the hard-coded default value '50 ms'

mck1117, 2022-09-14:
I would also support a change to `hz` instead of `ms`. TS lets you pick hz, so why don't we?

rusefillc, 2022-09-14:
we can switch logic from ms to hz and keep config location, effectively migrating everyone from 50ms to 50hz :)

too nasty?

mck1117, 2022-09-14:
evil, I like it

PR #4837 (2022-11-27): SD logging period -> frequency
I decided to go ahead and make this a config break, since the user may have had some other value set there that produces unpredictable results when converted from ms to hz.

fix #4553