rusefi/rusefi #4556: RTC and SD logs

rusefillc, 2022-09-09:
Does RTC and SD names work as of today? Proteus? Hellen?

We have a report https://rusefi.com/forum/viewtopic.php?f=19&t=2428

rusefillc, 2022-09-14:
I've just seen proper file names with date on hellen154h

mck1117, 2022-09-14:
> I've just seen proper SD card file names with date on hellen154h

are they correct?

ie,
1. install battery, connect ECU to rusefi console so it sets time
2. unplug ECU, leave on shelf unpowered for 24h
3. power without USB so that it writes a log
4. check timestamp, confirm it's right

rusefillc, 2022-09-14:
I love the test plan and have no time to try :(

mck1117, 2022-09-14:
really? it takes 5 minutes twice, 24 hours apart

mck1117, 2022-09-14:
@BeerMoneyMotorsports since @rusefillc seems unable, can you give it a try on latest prerelease firmware?

BeerMoneyMotorsports, 2022-09-15:
I will try it tomorrow morning. I just updated an ecu that i fixed. Set timestamp and all. Ill run it in the lemon.

BeerMoneyMotorsports, 2022-09-15:
it does not maintain correct time.  i just did a log and it still shows the time from when i connected console last night.

mck1117, 2022-09-15:
so that's a no, it doesn't work

mck1117, 2022-09-20:
```
#define STM32_LSE_ENABLED                   FALSE
```
seems like a smoking gun to me

rusefillc, 2022-09-20:
Actually
```
// todo: apply LSE patch
```

rusefillc, 2022-09-20:
#463 says 'hello 2022'

mck1117, 2022-09-20:
well, we shouldn't technically need LSE for the rtc to do *something*

LSI will still work, if inaccurate. I believe hellen boards have an LSE, though.

mck1117, 2022-09-20:
rtc survives soft reset (no power cycle) with LSI as-is, need to test with battery

rusefillc, 2022-12-25:
@nmschulte any interest to poke further? Any interest to restore LSE auto-detection see https://github.com/rusefi/rusefi/issues/463?

nmschulte, 2023-01-06:
I haven't built or tested the changes for ChibiOS or rusEFI (no LSE hardware).

I added logic for all STM32 devices in ChibiOS, and re-enabled LSE for the three STM32 `mcuconf.h` (unsure which ones need/want it).

I didn't re-add the `rtcWorks` sentinel flag, figuring this can be checked via ChibiOS mechanisms i.e. `(RCC->BDCR & RCC_BDCR_LSERDY) != 0` (depending specific STM32 variant).

rusefillc, 2024-08-29:
See https://github.com/FOME-Tech/fome-fw/issues/80

rusefillc, 2024-08-29:
```
// Allows LSE init to timeout and configure fallback RTC clock source in case
#define RUSEFI_STM32_LSE_WAIT_MAX           1000000
#define RUSEFI_STM32_LSE_WAIT_MAX_RTCSEL    STM32_RTCSEL_LSI
```

rusefillc, 2024-08-29:
```
// see RUSEFI_STM32_LSE_WAIT_MAX_RTCSEL
#define STM32_RTCSEL                        STM32_RTCSEL_LSE
```

larsnergard, 2024-09-11:
Can confirm RTC is not working on Uaefi build Nightly 070924. And it does not survive soft reboot either.

dron0gus, 2025-01-07:
https://github.com/rusefi/ChibiOS/pull/52

dron0gus, 2025-01-07:
https://github.com/rusefi/rusefi/pull/7262

dron0gus, 2025-01-07:
[image]
Seems LSI is unusable for RTC application.

nmschulte, 2025-01-08:
This matches my suspicions about the issue as well -- thanks for fixing up the ChibiOS patches :).

> ***nmschulte** — 8/15/24, 8:40 AM*
> > I wonder if the ChibiOS RTC LSE->LSI `WAIT_MAX` logic isn't working. 

[image]

I also noted this bit about STM32 H7 in ChibiOS:
> ***nmschulte** — 8/15/24, 7:11 PM*
> > was just looking at the backup domain SRAM clock enable logic and it's goofy/unaccounted for (or, generally/non-specifically accounted for -- perhaps H7 uses different terms/technique)

----

> Seems LSI is unusable for RTC application.

Would be interesting to see what practical values are actually; if there's a way to report/benchmark this through current boards or so (via compare w/ external osc or PC timing, etc.).

dron0gus, 2025-01-08:
Proteus v0.7 F4 after night with no external power supply, but with CR2032 battery. First connected to wall charger for few minutes that to PC:
```
$ date
Wed 08 Jan 2025 10:58:07 AM MSK
$ ls /media/dron/disk/ -l
total 1376
-rw-r--r-- 1 dron dron       2 Jan  8 04:17 index.txt
-rw-r--r-- 1 dron dron 1372160 Jan  8 04:19 re_250108_011758.mlg
```
Clock is retained but not ticking?
Console connected:
``
2025-01-08_11_00_17_412: EngineState: Current RTC time: 08 Jan 2025 01:19:42
...
2025-01-08_00_26_46_014: EngineState: STM32_RTCSEL=1 LSE actual=2 LSI
2025-01-08_00_26_46_015: EngineState: RUSEFI_STM32_LSE_WAIT_MAX_RTCSEL=2 LSI
...
2025-01-08_11_00_17_524: EngineState: Current RTC time: 08 Jan 2025 01:19:43
2025-01-08_11_00_17_524: EngineState: Time is changed to
2025-01-08_11_00_17_525: EngineState: Current RTC time: 08 Jan 2025 11:00:17
```

dron0gus, 2025-01-08:
LSI clock, Proteus is constantly powered from USB:
```
2025-01-08_11_42_46_897: EngineState: Current RTC time: 08 Jan 2025 11:42:37
...
2025-01-08_11_49_27_710: EngineState: Current RTC time: 08 Jan 2025 11:48:53
```
From -9 sec to -36 sec drift within 6 minutes. Should we implement LSI calibration (waste of time) or spend few cents for mandatory LSE 32768Hz oscillator?

zed65, 2025-01-09:
I have 32.768kHz crystal on my disco-407 board. Seem like the time is retained but not ticking when unplugged with only cr1220 battery

dron0gus, 2025-01-09:
@zed65 can you share console log with current master fw?

zed65, 2025-01-09:
[time1.txt](https://github.com/user-attachments/files/18364377/time1.txt)