rusefi/rusefi #4557: Proteus with SD card outside of vehicle looses a second

rusefillc, 2022-09-10:
master has seconds watchdog behind special tempBooleanForVerySpecialLogic setting

I have F7 proteus with SD card hooked to power supply fatal im under 15 minutes
I have MRE without SD card survive 20 minutes

[image]

rusefillc, 2022-09-10:
[image]

rusefillc, 2022-09-10:
[image]

rusefillc, 2022-09-10:
@mck1117 @dron0gus fun fact - MRE without SD card has survived the night

let me check with lower SD card period

rusefillc, 2022-09-10:
[image]

rusefillc, 2022-09-10:
@racer-coder this _might_ be related to

```
struct WrapAround62 {
	uint64_t update(uint32_t source) {
		// Shift cannot be 31, as we wouldn't be able to tell if time is moving forward or
		// backward relative to m_upper.  We do need to handle both directions as our
		// "thread" can be racing with other "threads" in sampling stamp and updating
		// m_upper.
		constexpr unsigned shift = 30;

		uint32_t upper = m_upper;
		uint32_t relative_unsigned = source - (upper << shift);
		upper += int32_t(relative_unsigned) >> shift;
		m_upper = upper;

		// Yes we could just do upper<<shift, but then the result would span both halves of
		// the 64-bit result.  Doing it this way means we only operate on one half at a
		// time.  Source will supply those bits anyways, so we don't need them from
		// upper...
		return (efitick_t(upper >> (32 - shift)) << 32) | source;
	}
```

rusefillc, 2022-09-10:
[image]

rusefillc, 2022-09-10:
[image]

rusefillc, 2022-09-10:
[image]

racer-coder, 2022-09-10:
Well that's interesting since I thought your tick clock was around 4MHz?

dron0gus, 2022-09-11:
tempBooleanForVerySpecialLogic code do uint32_t (efitimems_t) to float conversion.
It can cause precision lose when crossing 24 bit boundary
https://stackoverflow.com/questions/68690961/why-does-converting-int-to-float-lose-precision-in-c-programming-language
Can we remove this conversion and re-run test? 
https://github.com/rusefi/rusefi/pull/4562

rusefillc, 2022-09-12:
@dron0gus first time not 1100ms

[image]

dron0gus, 2022-09-12:
Half night of running with https://github.com/rusefi/rusefi/pull/4567 on my HW.
[image]

rusefillc, 2022-09-12:
2 out of (4 proteus and 1 mre) got affected in 8 hours

[image]

[image]

dron0gus, 2022-09-13:
@mck1117 
[image]
[2022-09-13_22.18.17.zip](https://github.com/rusefi/rusefi/files/9560247/2022-09-13_22.18.17.zip)

mck1117, 2022-09-13:
Man 40ms is a long lock.

rusefillc, 2022-09-13:
Last long lock I remember was engine sniffer or text output memory buffer copy

rusefillc, 2022-09-13:
Also 40ms is more for #4570 but really not for 1125ms problem?

mck1117, 2022-09-13:
That field can only store 65.535ms, so it could very well be 1125ms problem.

rusefillc, 2022-09-14:
Fun fact: I have three units not fatal with tempBooleanForVerySpecialLogic for 24 hours

I reset those and all three get fatal within 3 hours

mck1117, 2022-09-14:
Here are two logs that might be of interest. 

[blip.zip](https://github.com/rusefi/rusefi/files/9562672/blip.zip)

`re_31 scott.mlg` it from rusEFI proteus
`rc_15 scott.log` is from a racecapture connected over CAN

Have a look at 2007.52 seconds in the racecapture log. ECU data freezes for ~1.1 seconds.

In the rusEFI log, the same instant is 2010.97 seconds, where the log skips 1.148 seconds.

The ECU didn't blip (and time skip forward), the ECU hung for 1.1 seconds, then "snapped out of it" and went back to normal. The engine continued running for another hour without an issue.

mck1117, 2022-09-14:
WE GOT ONEEEEEEEE

[image]

[image]

4580649 / 4e6 = 1.145 seconds

mck1117, 2022-09-14:
Two while loops here without a timeout

[image]

mck1117, 2022-09-14:
Caught another one with some counters added to detect stuck loops.

Second loop (`STM32_RTC_HAS_SUBSECONDS`) takes 2 million iterations to complete.

[image]

mck1117, 2022-09-14:
@dron0gus what's your take? It looks like they're trying to exactly synchronize to a change of the subsecond counter. I don't understand why they're doing it under lock. That seems like a huge mistake in a real time system. Why do they not lock a mutex, synchronize clock, then unlock mutex?

dron0gus, 2022-09-14:
@mck1117 I think there is logic problem.
`oldssr` should be update on each iteration. It should be inside `do {} while` loop.
We are good with `tr`, `dr`, `cr` reading if `oldssr` (that we sampled right before reading `tr`) is the same as after finishing reading this batch of registers.
Otherwise if `ssr` ticked while we were reading other register code will stuck until `ssr` makes wrap around.

dron0gus, 2022-09-14:
Actually there are two modes of reading these RTC registers. Direct and shadowed. Shadow mode is used if `BYPSHAD != 1`.
By default `BYPSHAD` seems is cleared.
So there is no need do multiple reads and compare values.
[image]
But this code seems implements all checks for both BYPSHAD = 0 and BYPSHAD = 1 mode.

dron0gus, 2022-09-14:
Funny thing. This code was added as "fix" of some other bug.
https://sourceforge.net/p/chibios/bugs/1231/

```
commit 1a8ae11322a6ae870bc2299e80d410a8ba0dbe1e
Author: Giovanni Di Sirio <gdisirio@gmail.com>
Date:   Sat Apr 9 07:54:26 2022 +0000

    Fixed bug #1231.
    
    git-svn-id: svn://svn.code.sf.net/p/chibios/svn/branches/stable_20.3.x@15576 27425a3e-05d8-49a3-a47f-9c15f0e5edd8

diff --git a/os/hal/ports/STM32/LLD/RTCv2/hal_rtc_lld.c b/os/hal/ports/STM32/LLD/RTCv2/hal_rtc_lld.c
index 3c0632ce2..ebd10ddbe 100644
--- a/os/hal/ports/STM32/LLD/RTCv2/hal_rtc_lld.c
+++ b/os/hal/ports/STM32/LLD/RTCv2/hal_rtc_lld.c
@@ -637,7 +637,7 @@ void rtc_lld_get_time(RTCDriver *rtcp, RTCDateTime *timespec) {
   uint32_t dr, tr, cr;
   uint32_t subs;
 #if STM32_RTC_HAS_SUBSECONDS
-  uint32_t ssr;
+  uint32_t oldssr, ssr;^M
 #endif /* STM32_RTC_HAS_SUBSECONDS */
   syssts_t sts;
 
@@ -649,11 +649,17 @@ void rtc_lld_get_time(RTCDriver *rtcp, RTCDateTime *timespec) {
   while ((rtcp->rtc->ISR & RTC_ISR_RSF) == 0)
     ;
 #if STM32_RTC_HAS_SUBSECONDS
-  ssr = rtcp->rtc->SSR;
+  oldssr = rtcp->rtc->SSR;^M
+  do^M
+#endif /* STM32_RTC_HAS_SUBSECONDS */^M
+  {^M
+    tr  = rtcp->rtc->TR;^M
+    dr  = rtcp->rtc->DR;^M
+    cr  = rtcp->rtc->CR;^M
+  }^M
+#if STM32_RTC_HAS_SUBSECONDS^M
+  while (oldssr != (ssr = rtcp->rtc->SSR));^M
 #endif /* STM32_RTC_HAS_SUBSECONDS */
-  tr  = rtcp->rtc->TR;
-  dr  = rtcp->rtc->DR;
-  cr  = rtcp->rtc->CR;
   rtcp->rtc->ISR &= ~RTC_ISR_RSF;
 
   /* Leaving a reentrant critical zone.*/

```

dron0gus, 2022-09-14:
While code initially suggested by user in https://forum.chibios.org/viewtopic.php?f=35&t=5949 thread is correct and ssr is readed inside the loop.

dron0gus, 2022-09-14:
And yes, I see not reason to lock this code.

rusefillc, 2022-09-14:
I've reported https://sourceforge.net/p/chibios/bugs/1234/

closed 2022-09-14 by rusefillc

rusefillc, 2022-09-14:
So we are affected since https://github.com/rusefi/ChibiOS/commit/eed346d57f80efd78bd4130c7088066a1169d495#diff-477173e15cc7667e590339b9db05629c3cce0aa7956b5baaecb2bff2adcdd155

rusefillc, 2022-09-17:
@dron0gus @mck1117 see https://forum.chibios.org/viewtopic.php?f=35&t=5949&e=1&view=unread#p41462