rusefi/rusefi #4559: Request: IAC adder for power steering

10AE0739, 2022-09-10:
Miatas have an input from the power steering pump that tells the factory ECU to increase the IAC valve to compensate for the extra load. Like AC and fans, I’d like to use this input to tell the ECU to increase IAC valve.

rusefillc, 2022-09-10:
quick hack to test would be lua

something like https://github.com/rusefi/rusefi/wiki/Lua-Scripting#getauxanalogindex combined with
https://github.com/rusefi/rusefi/wiki/Lua-Scripting#setidleadd

if/once Lua prooves that's beneficial we can add as hard-coded logic

Do you need help to Lua this?

10AE0739, 2022-09-10:
Great idea! I've never used Lua but I know C and several other languages, so I can pick this one up quickly. I do need help with how I'd used getauxanalogindex to get input from the power steering pump. It is pin 1G - Steering/RES2.