rusefi/rusefi #4584: Compiled Proteus F4 don't work

hugovw1976, 2022-09-16:
I made a custom board (Proteus F4 pin to pin compatible) about a year ago and start to compile firmware just to eliminate some can bus messages for haltech dash (I have some board that send 4 wheel speed data, outside temp, gear shift possition, etc.) . Since then all work fine (I update my repositorys wiht new realeases and have no problem with compiled binaries) but when August 2022 Release - "Day 170" came out, when I compile (compile_protes_f4.sh) my board don't work, wen connect usb to my pc don't recognize any device and only power led is on, I clone a new clean repository without any changes just to be sure it's not the changes I made the problem but have the same problem.  I compile the compile_proteus_f4_hardware_QC_special_build and this work fine, pc recognize device when plug board and all outputs run in sequence, so I have no clue why compiled file for: compile_proteus_f4_hardware_QC_special_build work fine but: compile_proteus_f4 don't work. And same with new release (September 2022 Release - "Day 203")
Any advice wath can be wrong is apreciate.

rusefillc, 2022-09-27:
@mck1117 shall we ask any questions or close or else?

rusefillc, 2022-09-27:
@mck1117 would it help to have a pice of that HW at your place in Washington, USA?

mck1117, 2022-09-27:
> @mck1117 would it help to have a pice of that HW at your place in Washington, USA?

if the release binary works, then no, I don't need to touch this hardware.

mck1117, 2022-09-27:
@hugovw1976 what environment do you use? can you run `uname -a`?

what compiler do you use? can you run `arm-none-eabi-g++ -v`?

hugovw1976, 2022-09-28:
Windows 10 Home 21H2, GCC Compiler
`$ uname -a: 
CYGWIN_NT-10.0-19044 MiPC 3.3.5-341.x86_64 2022-05-13 12:27 UTC x86_64 Cygwin
`
`$ arm-none-eabi-g++ -v: 
gcc version 10.3.1 20210824 (release) (GNU Arm Embedded Toolchain 10.3-2021.10)`
Also try on virtual Linux Mint installing all needed with "setup_linux_environment.sh" but I have the same issue.
On Linux:
`uname -a : 
Linux r43bug-Virtual 5.15.0.47-generic #51-ubuntu SMP Thu Aug 11 07:51:15 UTC 2022 x86 64 x86 64 GNU/Linux`

`arm-none-eabi-g++ -v: 
gcc version 10.3.1 20210621 (release) (15:10.3-2021.07-4)`

mck1117, 2022-09-28:
gcc 10 is currently not supported, please use the official `setup_linux_environment.sh` script to configure gcc 9.3

hugovw1976, 2022-09-28:
Ok I check on weekend rigth now I'm busy.

commit 84ce8a89 (2022-09-28): GCC10 should be prohibited until it generates a functioning binary fix #4584
commit cab3a635 (2022-09-28): GCC10 should be prohibited until it generates a functioning binary fix #4584
closed 2022-09-28 by rusefillc