rusefi/rusefi #4603: LTS: signature considerations of long time support branches

rusefillc, 2022-09-22:
There is a desire to add Long Time Support branches model

We would need a way to recognize that build is from specific branch, we need to adjust signature logic somehow.

Reminder that signature is public API for tunerstudio .ini integration, they produce URL from current signature format

maybe DD is not required to be numeric? maybe we can replace YYYY.MM.DD with YYYY.MM.rX where bX is r0 r1 r2 etc for each branch build

chuckwagoncomputing, 2022-09-22:
Why would DD need to be replaced? Why not just add the branch to the signature, e.g. 2022.09.22.master.all.1303908112

rusefillc, 2022-09-22:
DD does not need to be replaced, your proposal is a reasonable alternative 

In my proposal yyyymm were branch name, in your proposal we need some sort of branch name file or other mechanism to lookup branch names. Works for me.

mck1117, 2022-09-22:
Do we know exactly what logic TS uses to map signature -> URL? We need to ask Phil that question before we change the format.

rusefillc, 2022-09-22:
We can make a pretty compatible signature and just test

chuckwagoncomputing, 2022-09-22:
Is there any reason https://github.com/rusefi/rusefi/blob/master/firmware/gen_signature.sh couldn't call `git` for the branch?

mck1117, 2022-09-22:
> Current branch names are looooong

So what? we don't actually build anything off a branch today. We could easily use branches like `lts-2209` for example, which is nice and short.

rusefillc, 2022-11-30:
@mck1117 my recollection is that back in September you were planning to contribute to LTS.

Is this a legit recollection or is it just in my head?

chuckwagoncomputing, 2023-01-21:
Is this completed with #4950?

rusefillc, 2023-01-21:
yes i hope so!

closed 2023-01-21 by rusefillc