rusefi/rusefi #4604: LTS: artifact upload change

rusefillc, 2022-09-22:
If we LTS we shall upload artifacts not into root folder of https://rusefi.com/build_server/ but a subfolder for LTS builds?

rusefillc, 2022-10-27:
@mck1117 my recollection of the phone conversation was that you were very much interested to start making progress. Any updates?

chuckwagoncomputing, 2023-01-21:
Is this completed with #4950? My understanding is that TS's "firmware fetch" mechanism defines where we put the firmware on the server. So unless we request that TS handle LTS builds differently, this has been decided for us.

rusefillc, 2023-01-21:
Not done - we still upload all snapshots into same folder, we have no folder structure for release branch snapshot uploads

https://github.com/rusefi/rusefi/pull/4950 was about TS downloading .ini

this ticket is about bash uploading .zip

chuckwagoncomputing, 2023-01-21:
Oh yes I'm an idiot.
Would each LTS build replace the last, or do we need to keep n LTS versions?

rusefillc, 2023-01-21:
For the purposes of this ticket it's definitely only one latest per branch, i.e. here we move from "only latest, one branch" to "only latest, multiple branches"

rusefillc, 2023-01-22:
@chuckwagoncomputing thank you for https://github.com/rusefi/rusefi/pull/4989 and https://github.com/rusefi/rusefi/pull/4990 this starts to look as something!

I've just created two test branches https://github.com/rusefi/rusefi/tree/lts-jan and https://github.com/rusefi/rusefi/tree/lts-2023.01

also created https://rusefi.com/build_server/lts-jan/ and hit the action

rusefillc, 2023-01-22:
@chuckwagoncomputing would it be possible to automatically create subfolder at https://rusefi.com/build_server/lts-jan/ for LTS? Any manual activity is just a chance for issues

chuckwagoncomputing, 2023-01-22:
Yes it's possible to automatically create folders.
Maybe I wasn't clear enough or haven't understood properly, but currently everything uploads to build_server/lts/, not build_server/<branch name>
If that's not what you meant, I can change that.

rusefillc, 2023-01-22:
My bad, https://rusefi.com/build_server/lts/lts-jan/ added as well

Please automatically create folders and I will test with https://github.com/rusefi/rusefi/tree/lts-2023.01

rusefillc, 2023-01-22:
Just created an empty https://github.com/rusefi/rusefi/wiki/Internal-Technical-Document-Long-Term-Support-LTS

chuckwagoncomputing, 2023-01-22:
#4991 will automatically create folders e.g. build-server/lts/lts-jan when running an lts build.
Just so we're clear - with this method of running LTS builds (the PR you merged) there is no requirement for the branch name. It doesn't have to start with 'lts' to work.

rusefillc, 2023-01-22:
let's see what workflow we end up with, i see us going to the automatic build with prefix approach or maybe conditional based on something like files or configuration. We would only know once we try

chuckwagoncomputing, 2023-01-22:
Oh I introduced a bug, now everything uploads to the lts directory.

closed 2023-01-28 by chuckwagoncomputing