rusefi/rusefi #4624: flip polarity of initializeSubaru_SVX

mck1117, 2022-09-26:
This trigger needs to be using sync mode `SyncEdge::RiseOnly`, but right now is configured for falling edges.

I think it just needs rise/fall swapped (and invert set in the tune), but not completely sure.

rusefillc, 2022-09-26:
What's the problem statement? What makes you suspect drama in that trigger?

mck1117, 2022-09-28:
This is prompted by #4621.

Before 4621, there were:

1.use only rise, sync on rising edges
2. use only _FALL_, sync on falling edge `<-- same as 1, but with invertPrimary=true`
3. use both edges, sync on falling
4. use both edges, sync on rise
5. use both edges, sync on both

After 4621, the options are
- use only rise, sync on rise
- use both, sync on fall
- use both, sync on rise
- use both, sync on both

This SVX trigger uses option 2 which is now redundant, so it's removed by 4621. Besides, it uses VR sensors, and the max9924 is invalid for falling edges anyway, right? @dron0gus what chip does svx board use for VR conditioning?

dron0gus, 2022-10-03:
SVX board uses MAX9924 for Crank and NCV1124 for Cam.
I have a plat to review SVX trigger and to divide it into two separate Crank and Cam patterns.
So I'll take a look.

mck1117, 2022-10-06:
Great, thanks! With the 9924 the only "valid" combination should be `SyncEdge::RiseOnly` with invert=false.