rusefi/rusefi #4626: Firmware Update Error

BeerMoneyMotorsports, 2022-09-28:
Unable to update firmware for Miata.  See errors below from NB1 and NA6.  I have deleted and reinstalled drivers with no success.  

**Console version 20220926
Windows 10.0
Bundle rusefi_bundle_hellen-nb1
Using selected COM10

Reboot command [reboot_dfu] sent into serial=COM10!

Giving time for USB enumeration...
Executing wmic path win32_pnpentity where "Caption like 'STM32 Bootloader'" get Caption,ConfigManagerErrorCode /format:list
No Instance(s) Available.

Done!

No Instance(s) Available.
*** DRIVER ERROR? *** Did you have a chance to try 'Install Drivers' button on top of rusEFI console start screen?**

[image]

mck1117, 2022-09-28:
snip of device manager?

BeerMoneyMotorsports, 2022-09-28:
[image]

[image]

BeerMoneyMotorsports, 2022-09-28:
Confirmed same issue on release for Hellen 121 Nissan.  STM cube programmer will connect while in DFU

[image]

rusefillc, 2022-09-28:
I've only tested #4625 on Win7 so far

Could be that they request elevation now?

I will have access to hardware in 25 hours from now.

BeerMoneyMotorsports, 2022-09-29:
Do you need any more data from my end?

rusefillc, 2022-09-29:
No

Wow, on win11 they have two spaces not one.

[image]

closed 2022-09-29 by rusefillc

rusefillc, 2022-09-29:
Looks good with the fix

[image]

commit a6a5b6fd (2022-09-29): Firmware Update Error fix #4626