rusefi/rusefi #4635: 4B11 trigger error, double RPM

ElDominio, 2022-10-05:
https://rusefi.com/online/view.php?log=1065

at 7.8 seconds RPM jumps to double RPM

rusefillc, 2022-10-05:
Double RPM is usually cam noise but this tune already has 3000 cam sync limit

https://rusefi.com/online/view.php?msq=1249&dialog=triggerConfiguration_IO

[image]

rusefillc, 2022-10-05:
camResyncCounter did not go up so that's NOT cam resync

Maybe it's legit trigger noise? Ideally root cause of noise should be addressed. I can add resiliency gap ratios into decoder but that would only hide the problem. 

Reminder about https://github.com/rusefi/rusefi/issues/4507 :(

[image]

ElDominio, 2022-10-05:
Any type of filtering would be nice, since the volume of these units will be with plug and play harness.

mck1117, 2022-10-05:
> I can add resiliency gap ratios into decoder but that would only hide the problem.

This won't help, the warning is that there were *too many* teeth. Also, look at instant RPM: it's fuzzy:

[image]

He's going to try inverting trigger, adjust offset, and see if it goes away (ie, some hall sensors are crap on one edge or the other)

ElDominio, 2022-10-06:
Fixed by setting cam to inverted.

mck1117, 2022-10-06:
@ElDominio, did the instant rpm get not-fuzzy?

mck1117, 2022-10-06:
That looks pretty fuzzy to me. Did you try inverting the primary trigger?

ElDominio, 2022-10-06:
Yes, that log is inverted.

ElDominio, 2022-10-07:
UPDATE:
Not fixed. I had set a max cam/crank sync RPM, which made it not lose sync. If Max RPM for sync = 0, it loses sync at about 5k rpm. Setting a max RPM for sync = 3000 lets it rev out without a problem, but VVT doesn't work (since it seems this value makes the ECU just completley forget about the existence of VVT)

Log with Max Sync RPM disabled (=0)
https://rusefi.com/online/view.php?log=1069

Log with max cam sync = 4000RPM
https://rusefi.com/online/view.php?log=1070

ElDominio, 2022-10-07:
Test user also reports this [image]

ElDominio, 2022-10-08:
[36-2-1log.zip](https://github.com/rusefi/rusefi/files/9740339/36-2-1log.zip)

logic analyzer log, our hand slipped a few times llol bit i can see the pattern

mck1117, 2022-10-08:
thank you, will review tomorrow on the plane!

ElDominio, 2022-10-11:
[image]

[2022-10-10_02.47.38.mlg.zip](https://github.com/rusefi/rusefi/files/9751568/2022-10-10_02.47.38.mlg.zip)

Hmm still looks fuzzy...

mck1117, 2022-10-11:
This part looks not-fuzzy to me (ie, good!)

[image]

mck1117, 2022-10-11:
This is a sync loss due to a cam resync - let's get a logic analyzer log of cam+crank while repeating the behavior in that log and see what we see.

[image]

ElDominio, 2024-02-23:
Uh, fixed? A bunch of 4Gxx's are running great