rusefi/rusefi #4649: MRE rev limit max 12500rpm

OrchardPerformance, 2022-10-09:
Was bench testing something earlier and found the max "rpm hard limit" possible to set on an MRE is 12500. 
Setting it to more has unpredictable results with the numbers in the box shortening to 4 digits (i.e. 1200 not 12600) 

This also seems to have the result of the ignition output getting stuck on if trigger stimulation is also activated. 

Going to test some more but this may be a limit across all RE. 

Relevance here is motorcycle could be 14k to 15k rpm.

OrchardPerformance, 2022-10-09:
Update - Tried MRE, Proteus and Frankenso and this exists on all of them. 
12000 is ok 
[image]

13000 changes to "200" 
[image]

14000 changes to "1200" 
[image]

Could this be a TS problem?

OrchardPerformance, 2022-10-09:
Looks like 12799 is the highest the box takes. 
12800 turns into zero. 

Seems like 128 x10 is the highest the number goes before overflow? 
12799 = 12799
12800 = 0
12801 = 1
14000 = 1200

rusefillc, 2022-10-09:
I see zero tune URLs

i see firmware version not being mentioned

I see https://github.com/rusefi/rusefi/pull/4300 being very relevant

rusefillc, 2022-10-09:
I also see 19000 locally

overall I tend to blame operator not TS

[image]

OrchardPerformance, 2022-10-09:
Version does not matter, I tried at least 7 versions going back as far as 2019. 

The problem exists on all versions where the RPM limit moved to the "limits and fallbacks" page. 

Tune totally irrelevant, see above. 

You can type any number you like, but soon as you press enter 19000 will change to 6200.

mck1117, 2022-10-09:
@rusefillc read the issue please. He didn't say you can't type in 19000, he said that values above 12800 behave like x-12800, not that they can't be typed in. Smells like the config field is an int8_t and we're running out.

rusefillc, 2022-10-09:
@mck1117 I appreciate your loyalty but in my opinion it's misplaced. Clearly emotional baggage is dragging us all down here but in this specific case I am still thinking that I am paying attention while you guys do not.

https://rusefi.com/online/view.php?msq=1256&dialog=limitsAndFallback

[image]

mck1117, 2022-10-09:
Then why didn't we start with the `git blame`? Looks like maybe old firmware here.

OrchardPerformance, 2022-10-09:
Ok, so it's fixed but the issue DID exist. 

I will inform any users of exiting units with firmware that may not be the latest that this bug exists and can be fixed by updating. 

This ticket at least now exists to catalogue what was a bug and can be found by anyone that also encounters it.

closed 2022-10-09 by OrchardPerformance

rusefillc, 2022-10-09:
Just a reminder that I've literally referenced #4300 fix

I am hoping for @OrchardPerformance to see signal here. Screenshots bad urls good maybe?

mck1117, 2022-10-09:
there was even already an issue: #4298