rusefi/rusefi #4651: efiPrintf does not accept some special chars

dron0gus, 2022-10-10:
Message parser on PC side confused by '\r' or '\n' (or both). And console skips some following messages.
```
I 221010 115627.167 [communication executor1] MessagesCentral - postMessage EngineState: Sleeping...
I 221010 115628.174 [communication executor1] MessagesCentral - postMessage EngineState: Restarting at 115200
I 221010 115628.177 [communication executor1] MessagesCentral - postMessage EngineState: sending AT
I 221010 115628.177 [communication executor1] EngineState - EngineState.unknown: `msg`Received 5 +OK
I 221010 115628.178 [communication executor1] EngineState - Invalid key [] value [`msg]
I 221010 115628.178 [communication executor1] EngineState - EngineState.unknown: Received 5 +OK
I 221010 115628.178 [communication executor1] EngineState - EngineState.unknown: `msg`sending AT+VERSION
I 221010 115628.178 [communication executor1] EngineState - Invalid key [] value [`msg]
I 221010 115628.178 [communication executor1] EngineState - EngineState.unknown: sending AT+VERSION
I 221010 115628.179 [communication executor1] EngineState - EngineState.unknown: `msg`Received 5 +VERS`msg`command AT+VERSION
I 221010 115628.179 [communication executor1] EngineState - Invalid key [] value [`msg]
I 221010 115628.179 [communication executor1] EngineState - EngineState.unknown: Received 5 +VERS`msg`command AT+VERSION
I 221010 115628.179 [communication executor1] EngineState - Invalid key [Received 5 +VERS] value [`msg]
I 221010 115628.179 [communication executor1] EngineState - EngineState.unknown: command AT+VERSION
I 221010 115628.179 [communication executor1] EngineState - EngineState.unknown:  failed -1`msg`FAIL! Only 0 commands (of 4 total) were passed to the Bluetooth module!`msg`Finished with -4 return`msg`UART: TS: ERROR: no second byte`msg`TunerStudio size=756 / total=653 / errors=1 / H=3 / O=313 / P=0 / B=0`msg`TunerStudio W=0 / C=0 / P=0`
I 221010 115628.180 [communication executor1] EngineState - Invalid key [ failed -1] value [`msg]
I 221010 115628.180 [communication executor1] EngineState - EngineState.unknown: FAIL! Only 0 commands (of 4 total) were passed to the Bluetooth module!`msg`Finished with -4 return`msg`UART: TS: ERROR: no second byte`msg`TunerStudio size=756 / total=653 / errors=1 / H=3 / O=313 / P=0 / B=0`msg`TunerStudio W=0 / C=0 / P=0`
I 221010 115628.180 [communication executor1] EngineState - Invalid key [FAIL! Only 0 commands (of 4 total) were passed to the Bluetooth module!] value [`msg]
I 221010 115628.180 [communication executor1] EngineState - EngineState.unknown: Finished with -4 return`msg`UART: TS: ERROR: no second byte`msg`TunerStudio size=756 / total=653 / errors=1 / H=3 / O=313 / P=0 / B=0`msg`TunerStudio W=0 / C=0 / P=0`
I 221010 115628.180 [communication executor1] EngineState - Invalid key [Finished with -4 return] value [`msg]
I 221010 115628.180 [communication executor1] EngineState - EngineState.unknown: UART: TS: ERROR: no second byte`msg`TunerStudio size=756 / total=653 / errors=1 / H=3 / O=313 / P=0 / B=0`msg`TunerStudio W=0 / C=0 / P=0`
I 221010 115628.181 [communication executor1] EngineState - Invalid key [UART: TS: ERROR: no second byte] value [`msg]
I 221010 115628.181 [communication executor1] EngineState - EngineState.unknown: TunerStudio size=756 / total=653 / errors=1 / H=3 / O=313 / P=0 / B=0`msg`TunerStudio W=0 / C=0 / P=0`
I 221010 115628.181 [communication executor1] EngineState - Invalid key [TunerStudio size=756 / total=653 / errors=1 / H=3 / O=313 / P=0 / B=0] value [`msg]
I 221010 115628.181 [communication executor1] EngineState - EngineState.unknown: TunerStudio W=0 / C=0 / P=0`
```

mck1117, 2022-10-10:
Should we instead sanitize those spots where we know this can happen, instead of taking the perf hit everywhere we print text?

dron0gus, 2022-10-27:
Now all non-printable chars are filtered out with `isprint`
Do we need to filter out back apostrophe?