rusefi/rusefi #4662: stop using unix time for console time setting

mck1117, 2022-10-13:
Touching the `time.h` unix time functions adds ~9k to the compiled binary size.  If we send a text date/time instead, we save that flash with no change in behavior.

nmschulte, 2022-11-09:
Is this in reference to the `hw_layer` `#include`s?  `firmware/console` does not:

`$ git grep '<time\.h>' firmware/`
```
firmware/controllers/algo/nmea.cpp:#include <time.h>
firmware/hw_layer/rtc_helper.cpp:#include <time.h>
firmware/hw_layer/sensors/gps_uart.cpp:#include <time.h>
```

mck1117, 2022-11-09:
Yes - it's a reference to those. The console (Java app on pc) sets the RTC on the ecu by sending a Unix timestamp - but the stuff on the ecu side to decode that ends up pulling in a huge pile of libraries. We could just send yyyymmddhhmmss instead which would require no decode.

nmschulte, 2022-11-09:
Alright.  Is this looking like the right place?  https://github.com/rusefi/rusefi/blob/master/java_console/io/src/main/java/com/rusefi/io/serial/RateCounter.java

rusefillc, 2022-11-09:
RateCounter? nooooo

I would go order a discovery board or nucleo board :)

```
#if EFI_RTC
	} else if (strEqualCaseInsensitive(paramStr, CMD_DATE)) {
		// rusEfi console invokes this method with timestamp in local timezone
		setDateTime(valueStr);
#endif
```

```
            if (ConnectionStatusLogic.INSTANCE.getValue() == ConnectionStatusValue.CONNECTED) {
                long unixGmtTime = System.currentTimeMillis() / 1000L;
                long withOffset = unixGmtTime + TimeZone.getDefault().getOffset(System.currentTimeMillis()) / 1000;
                consoleUI.uiContext.getLinkManager().execute(() -> consoleUI.uiContext.getCommandQueue().write(IoUtil.getSetCommand(Fields.CMD_DATE) +
                                " " + withOffset, CommandQueue.DEFAULT_TIMEOUT,
                        InvocationConfirmationListener.VOID, false));
            }

```

seems way more relevant. Somewhere in java

mck1117, 2022-11-09:
then corresponding `SetTimeUnixSec` and `GetUnixTimeSec` in c++

rusefillc, 2022-11-09:
nmea and gps_uart should probably be off with conditional compilation on most/all boards

rusefillc, 2022-12-03:
@nmschulte can you please test on real hardware?

nmschulte, 2022-12-03:
I'll try and test this evening, add some tests to the HW CI workflow if possible.

nmschulte, 2022-12-05:
```
I 221205 010126.945 [Commands Queue] BinaryProtocol - Sending [set date 2022-12-05T01:01:26.886]
I 221205 010126.986 [communication executor1] MessagesCentral - postMessage EngineState: date_set Date parameter 2022-12-05T01:01:26.886 is wrong
I 221205 010126.986 [communication executor1] MessagesCentral - postMessage EngineState: confirmation_set date 2022-12-05T01:01:26.886:32
```

rusefillc, 2022-12-07:
@nmschulte any luck with any test? I hear rumors that something is broken like a stack overflow?

nmschulte, 2022-12-07:
I am working another branch this evening to remove the `sscanf` dependency which bloats the stack -- I didn't conclude anything as far as a test: the code that was "failing" [gracefully] was the command handler in settings, so unless some logic is added to the simulator to simulate a command, or call the `setDateTime` method directly, I'm not sure what good a test there will do in this regard.

rusefillc, 2022-12-07:
@nmschulte do you have enough information to create a separate explicit ticket for simulator issue? Overall please consider keeping me in the loop

nmschulte, 2023-01-06:
I think this can be closed as complete?

closed 2023-01-06 by rusefillc