rusefi/rusefi #4665: Enable ethernet on one or both nucleo boards

rusefillc, 2022-10-14:
We have https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/nucleo_f767 and https://github.com/rusefi/rusefi/tree/master/firmware/config/boards/nucleo_h743

One day both should have ethernet enabled

ggurov, 2024-02-26:
Ethernet seems to be enabled, testing 2024-02-26 latest main
board is a Nucleo-F767ZIT6U (https://www.digikey.com/en/products/detail/stmicroelectronics/NUCLEO-F767ZI/6004740)

have to adjust the ipaddress, as there's no dhcp:
```

#define LWIP_IPADDR(p)            IP4_ADDR(p, 192, 168, 1, 4)
#define LWIP_GATEWAY(p)           IP4_ADDR(p, 192, 168, 1, 1)
```

in console/lwipopts.h

tunerstudio connects using wifi protocol with the right ip specified.

dron0gus, 2026-04-15:
Tested on F767:

[image]

Currently hardcoded IP 192.168.10.1, port 29000

rusefillc, 2026-04-15:
```
IP4_ADDR(p, 192, 168, 1, 10)
```