rusefi/rusefi #4670: Secondary TS port for Hellen128

shadowm60, 2022-10-16:
PE0 and PE1 is available on the PCB to use it as serial.

Use case would be connection of an ESP-01 running esp-link to have serial to wireless connection.
Tested with f4 disco i gor a data rate of around 7.4, this would be usable in the car, the only issue is i do not know how to configure it for UART8 that is located on the used pins.

Currently this is what i managed to poke around, but i am guessing there should be a better way:
https://github.com/shadowm60/rusefi/tree/hellen128_TS_Secondary

rusefillc, 2022-10-16:
Funny thing I've spend whole yesterday wondering why data rates are so lame in different use-cases

https://github.com/shadowm60/rusefi/commit/d67c3cdd152b9d25ceefaaf2dbdd6d087c0c62ba seems to touch a lot of not-128 files

also SD8 reminder

[image]

shadowm60, 2022-10-16:
Yes, i have touched them, moved a common think, that is not final, i'll find a way to make it better.

Uart8 on f429 is available, i need to figure out the proper setup, at least i am in a state that i hope it will work :))

Ecu now in the car with latest rel and esp in the metal housing, signal looks nice[image]
[image]

If anybody has inputs i am open for ideeas. Or maybe tickets that configured secondary, just a plain search did not work as i was expecting it to work